cartographer-project/cartographer_ros #1230: Configuration difference between provide_odom_frame and use_odometry

gaoethan, 2019-03-20:
There are several cartographer option which relates to odom, but I'm really not get it for the doc explanation for the option `provide_odom_frame` and `use_odometry`, but what's their real difference ?

BTW, from the doc of [configuration](https://google-cartographer-ros.readthedocs.io/en/latest/configuration.html), it seems that the option `odom_frame` and `published_frame` also have something to do with the odometry, can someone give some more direct and clear explanation?
Thanks.

ojura, 2019-03-20:
Published frame: the child frame of the map frame when publishing the robot location (which is the complete map->tracking frame transform).  If your tracking frame has no parents, you specify it directly as the published frame. If it does already have a parent, you specify the topmost frame with no parents (this is usually the case when you already have an odom frame as a parent to the tracking frame. You specify odom as the published frame.) 
Perhaps this setting would be more clear if called “map child frame”.

Use_odometry: whether to use the odom topic in slam as prior. Completely unrelated to tf.

Provide odometry: if you want smooth odometry-like localization in your tf tree (smooth, but drifting), you can have cartographer provide one for you. “Odom_frame” is the frame in which such odometry is published. You also need to adjust the published frame above (the map child frame) as well.

gaoethan, 2019-03-21:
>Provide odometry: if you want smooth odometry-like localization in your tf tree (smooth, but drifting), you can have cartographer provide one for you.

How does cartographer provide one? I mean, what does cartographer base on to calculate that one for the user?

If `use_odometry` is totally unrelated to tf, how does the cartographer use it internally? thanks !

ojura, 2019-03-21:
It’s the unoptimized poses (without loop closure), basically lidar odometry. These are smooth, but drifting.

The odometry topic contains odometry measurements as nav_msgs/Odometry messages, which contain an odometry pose and twist velocity (unused). The odometry pose, if not given with respect to the tracking frame, is transformed using the tf tree (usually static robot transforms) to the tracking frame. It is then used as a prior for scan matching in slam.

gaoethan, 2019-03-22:
Okay. I see, if `use_odometry` and `provide_odom_frame` are both set to true, there will be two `odom`, one is estimated on the lidar moving and the other is from other part of the system such as  the wheel etc.

One more question, `the tracking frame is the frame that is tracked by the SLAM algorithm`, for example, an IMU, generally speaking, how to grasp the `is tracked by the SLAM algorithm`? And will be sensor fusion with Lidar to estimate the odom while `provide_odom_frame` is enabled?

closed 2019-04-02 by gaoethan

ojura, 2019-04-02:
Sorry, I could not quite understand your last question. "is tracked by SLAM" means that SLAM is tracking the pose of that frame on the vehicle.