cartographer-project/cartographer_ros #1236: Problems with tuning

Semih67, 2019-03-26:
I'm using cartographer 3D, a pico flexx camera

I do not get good results of my map so here are my files:

lua:
https://www.dropbox.com/s/8k5dh7sycpzdyuy/emili_3d.lua?dl=0

urdf:
https://www.dropbox.com/s/ss91gpkjcpyu4kr/emili.urdf?dl=0

launch:
https://www.dropbox.com/s/1mimlpay7ziu0ze/cartographer_3d.launch?dl=0

bag:
https://www.dropbox.com/s/eky13q2t4i5qd7d/bag_issue.bag?dl=0

rosbag validate:
https://www.dropbox.com/s/fnp33sjf3deaxjm/rosbag_validate%20for%20bag_issues?dl=0

StefanXR, 2020-06-27:
@MrGitGo Did you manage to run the pico flexx camera with catographer? 
I have a pico flexx too. I am trying to map with catographer, but it couln't build a submap.

MrGitGo, 2020-06-28:
You cant get a good quality with rgbd cameras, try it with laser sensors instead

closed 2020-10-27 by MichaelGrupp