cartographer-project/cartographer_ros #1239: Unstable map->odom frame when using odometry

dnikitop, 2019-03-26:
Using localization only mode to localize robot in mapped area. Inputting reliable/smooth odometry data but see a very unstable map->odom tf frame. Mostly occurs when turning. 
The map->odom tf is the one shaking in the gif below https://giphy.com/gifs/8BkIhqzLYYnbCz6sgi 

Lowering the lidar sampling rate causes an extremely unstable frame which seems to be dragged by the odometry before it is reset by the scan. 
https://giphy.com/gifs/euCv4zqEIHuvD4cxr4

Is there an issue with the input of our odometry data?

1.  Rosbag validation - contains the output map_carto->odom tf.
https://gist.github.com/dnikitop/4c5c74fc15e37dd0b4602f2c123186f2

2. Using recent master release of cartographer and cartographer_ros 
Lua File Content: 
https://gist.github.com/dnikitop/4dd6d7f19e839d57d79d9c2bc5d454b2

3. Bag File of the robot turning in place with the resulting map_carto->odom frame:
https://drive.google.com/file/d/1szVYH1xUlkz0lPifBPnpo96cMPU3Wri0/view?usp=sharing

4. PBStream of the map we are trying to localize with: 
https://drive.google.com/file/d/1eJAbffkKt99Qif16ujLXPIYs3v5rpNdi/view?usp=sharing

remaps:
odometry  = /sensors/odometry/pose
lidar = /sensors/lidar/scan

tf tree:
We have a static tf pub from map -> map_carto. Cartographer is calculating map_carto->odom
map->map_carto->odom->base_footprint

MichaelGrupp, 2020-10-27:
Closing for inactivity, we can't invest time to help you with your setup anymore unfortunately.

closed 2020-10-27 by MichaelGrupp

thezest76, 2021-10-05:
Hello @dnikitop, I have a similar issue when I use cartographer for 2D mapping [This is the video link](https://www.youtube.com/watch?v=BAa7H98_aNA).  Not sure if my problem is the exact same as yours since your video links are not available anymore. I am just wondering if you have managed to fix your problem yet.

Best regards,

zixiao-bios, 2023-02-16:
Hello @han-kyung-min , I have the same situation as your video. Have you solved this problem？

han-kyung-min, 2023-02-16:
Hello, zixiao,

No I haven't. Unfortunately, the authors are not responding to my question.
:(