cartographer-project/cartographer_ros #1243: An error occurred while executing the "rosservice call /finish_trajectory 0"

zhaoyu24, 2019-03-27:
I run demo_backpack_2d.launch successfully.(_roslaunch cartographer_ros demo_backpack_2d.launch bag_filename:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag_
)
when i run "_rosservice call /finish_trajectory 0_" to save the map ,an error occured.
**ERROR: Unable to load type [cartographer_ros_msgs/FinishTrajectory].
Have you typed 'make' in [cartographer_ros_msgs]?**

I build and install cartographer_ros according to The official reference：https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html#building-installation
I want to know how to solve this problem or how to save map.

MichaelGrupp, 2019-03-27:
You probably didn't build `cartographer_ros_msgs` which is a sub package of `cartographer_ros`.

zhaoyu24, 2019-03-27:
Thanks for your response.Actually i know this ERROR means `cartographer_ros_msgs` didn't build,but i don't know why and i just follow official document.I would appreciate if you give me some advice about how to rebuild `cartographer_ros_msgs`.

MichaelGrupp, 2019-03-27:
I can't say more because I don't use the `catkin_make_isolated`commands written in the documentation (I use `catkin build`).

But maybe someone else or one of the maintainers can help.

zhaoyu24, 2019-05-08:
#1078 [(https://github.com/googlecartographer/cartographer_ros/issues/1078)]
this issue provides the solution.if you follow the official document to compiling cartographer_ros successfully,you need to do is `source` before calling service.

closed 2019-05-08 by zhaoyu24