cartographer-project/cartographer_ros #1247: odometry laserScan calibration

ZhengPan2014, 2019-03-29:
I want to know is that important to calibrate the base_link to laser frame accurately so that  I can use the odometry data for 2d mapping ? I try my best to install laserScan sensor in center of my turtlebot,  but unfortunatly with a small tranfrom T which was measured roughly by hand. The final map was messing and overlapping, even worse than mapping with only laserScan data.
anyone can give me some advise? appriciate for any references, thanks!
@esteve @SirVer @damienrg @damonkohler @skohlbr

Mafuqiang, 2019-05-29:
can you show me your urdf file with odometry? When I run cartographer with odometry, I get some error. Thank you very much.

Fanny-one, 2019-11-14:
Have you solve the issue? I have met the same problem.

MichaelGrupp, 2020-10-27:
We can't help with this calibration, but you need to make sure the scanner is level and the offset is correct. You could also check your odometry quality since you use it.

closed 2020-10-27 by MichaelGrupp

Alcatrazee, 2022-08-13:
you can actully use the lidar to do the mapping without the odometry, and tune the cartographer parameters, a well tuned cartographer slam is better than the one with a bad odometry, just a suggestion