cartographer-project/cartographer_ros #1250: How to use 2D cartographer for the navigation of a robot

Duke-Allen, 2019-03-30:
Hello, I have been running 2D cartographer successfully. Now I would like to use it in the navigation of my car. I have some questions about it. Could the cartographer use in the navigation? If so, how to do it in detail? I am new of the navigation, so I don't know how to do it. Any suggestions will be appreciation! 
Thanks a lot.

precyon, 2019-04-01:
@Duke-Allen, if you new to navigation you should go through the ROS navigation tutorial, http://wiki.ros.org/navigation/Tutorials, in detail (assuming you are using ROS). 

It is possible to have Cartographer give out pgm/yaml maps like gmapping/hectorslam in the above tutorial. Save the Cartographer state in your mapping stage as a pbstream file and run `cartographer_assets_writer` to give you those files. An example is already here: https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/launch/assets_writer_ros_map.launch

Duke-Allen, 2019-04-01:
@precyon , thanks for your advice. I will try it, thanks again.

joyde3382, 2019-04-17:
@Duke-Allen Did you have any success? I'm pretty much in the same situation as you, but can't seem to figure out a way to do it. Tried with navigation stack and move_base, which i've manage to get working by simulation. But can't seem to transfer it to my robot.

MichaelGrupp, 2020-10-16:
Closing as inactive.

closed 2020-10-16 by MichaelGrupp