cartographer-project/cartographer_ros #1253: Lookup would require extrapolation at time. When trying 2D Cartographer ROS with a YD Lidar

mahendranicl, 2019-04-06:
Hi,

I am trying to Run 2D Cartographer ROS with a YD Lidar. One error is ruining the performance of the move_base node, i.e. improper Robot navigation. 

It was working fine before in Ubuntu 14.04. When I upgraded the version to Ubuntu 16.04, I had to recompile my workspaces. 

Now when I run the 2D Cartographer ROS with a YD Lidar, I get the following error. 

[ WARN] [1554547766.837016521]: W0406 11:49:26.000000 19226 tf_bridge.cc:52] Lookup would require extrapolation at time 1554547766.814867700, but only time 1554547766.842371812 is in the buffer, when looking up transform from frame [laser_frame] to frame [base_link]

Thanks in advance !

c66tang, 2019-04-21:
Hi, I ran into the same issue. Any updates on this?

clement2449, 2019-05-14:
Hi, i have the same issue. It 's very anoying and there is nothing else about this problem on internet :(

clement2449, 2019-05-14:
[image]

mahendranicl, 2019-05-14:
Hi, 
I have solved the issue. It is to do with adjusting the pose update frequency, map update frequency, controller frequency parameters in both the move_base node configuration files and cartographer configuration files.

clement2449, 2019-05-14:
@mahendranicl i understand what you would say but i don't want to find the correct value :( can you give me your lua file please with the value you have change it will really help me i think

clement2449, 2019-05-14:
and your lidar file if you have change any value i am working on a lidar 7hz

mahendranicl, 2019-05-14:
There are more than 3-4 files involved from various nodes. Will try to send

clement2449, 2019-05-15:
@mahendranicl  thk you a lot i'm on it since 2 days ^^

artemiialessandrini, 2019-10-07:
Hi @mahendranicl 
Did I get you right, that solution is to increase a frequency of several topics published?

closed 2020-10-16 by MichaelGrupp