cartographer-project/cartographer_ros #1254: Localizing in 2D from a 3D map

ryanmeasel, 2019-04-06:
We are creating maps using VLP16s with cartographer in 3D configuration. Is it possible to localize in this map using a 2D planar lidar?

Further, can the map be updated while in localization mode?

shamengjun, 2019-09-05:
yes, cartographer support pure localization through load previously created .pbstream file(local slam state), meanwhile, it can't update submaps but lt can keep 3~5 submaps that scan match with previous submaps.

ryanmeasel, 2019-09-05:
@shamengjun Localization in 2D works from a pbstream created in 3D?

shamengjun, 2019-09-09:
@ryanmeasel I'm so sorry, this is not known by me currently, I am only focused on 2D currently.

MichaelGrupp, 2020-10-16:
This probably doesn't work because the 2D SLAM grid is completely different to the 3D structure that you saved.

closed 2020-10-16 by MichaelGrupp