cartographer-project/cartographer_ros #1255: Reset and load cartographer state during runtime of cartographer_node

shirbarzel, 2019-04-07:
Hey,
1. Is there a way to reset cartographer node state without restarting the process(during runtime)?
Maybe by publishing on some topic.
In reset I mean that the point at which the robot is a the moment will be (x=0,y=0) and it will start a new map(Just like starting the process without loading a pbstream file)

2. In addition I would like to know if there is a way to load state during run time? (loading a state from a pbstream file)

Thanks

MichaelGrupp, 2019-04-12:
I'm not entirely sure what you mean, but it could be related to something I want to implement:
Control over deleting and loading states during runtime, as proposed here: https://github.com/googlecartographer/rfcs/pull/41

shirbarzel, 2019-04-14:
Restart all data related to cartographer(Map, Landmarks, subsmap etc)  just like restarting the process of cartographer node, without loading a pbstream file

MichaelGrupp, 2020-10-27:
Completely resetting everything is currently not supported. My draft goes into this direction but is also not finished.
Regarding destruction, there was a similar request but it was not merged into master. You can follow up here: https://github.com/cartographer-project/cartographer/pull/1446

> Does the master version of cartographer is compatible with that version of cartographer_ros?

Nope. Only file backwards compatibility for pbstream.

closed 2020-10-27 by MichaelGrupp