cartographer-project/cartographer_ros #1256: Minimum hardware requirements

lukehutch, 2019-04-09:
The documentation says, re. hardware requirements:

> Although Cartographer may run on other systems, it is confirmed to be working on systems that meet the following requirements: 64-bit, modern CPU (e.g. 3rd generation i7); 16 GB RAM

However, this is not enough information when spec'ing out on-board computational hardware for a specific sensor, if Cartographer needs to run on-device. For example, approximately how much CPU power would be needed to run Cartographer in realtime with an Ouster OS-1^16 (327,680 points/sec) vs. OS-1^64 (1,310,720 points/sec) vs. OS-1^128 (2,621,440 points/sec) sensor, or for one vs. two Velodyne VLP-16 sensors?

gaschler, 2019-04-10:
The computational requirements depend on so many factors, not only on the sensors and CPU, so one cannot list system requirements.

Just to give you examples: 

- Parameters: When you change the voxel size and constraint search/loop closure search rate, your CPU load may easily change by an order of magnitude. In the docs, there is a section how to tune for speed/low latency: 
 https://google-cartographer-ros.readthedocs.io/en/latest/tuning.html#low-latency
- ROS: Most people use Cartographer with ROS, but you can also use it as a library. I hypothesize that the library is substantially more efficient at processing binary laser scans than the ROS node serializing and moving things around under much mutex locking.

The only thing I can say about CPU for sure is SSE/vector math is used heavily and more cores are better to find more loop closures. GPU is not used.

lukehutch, 2019-04-10:
@gaschler Fair enough. But I still have no idea of even what ballpack we're talking, within an order of magnitude or so. Can Cartographer be run in realtime on a Raspberry Pi, albeit with low snapping resolution etc.? (I'm guessing not.) What about a laptop or small form factor PC? (I'm guessing maybe, for a mid-to-high-end laptop, depending on the settings, but probably not on a low-end laptop.) Or should I only really consider capturing the data on a low-powered CPU for later offline processing on a high-end desktop or server?

gaschler, 2019-04-11:
3D algorithm: PCs can handle multiple lidars in realtime with plenty of loop closing. Again, the is leeway by setting parameters differently.

2D algorithm: Any recent x86 64 bit device, also small ones, should be fine. My PC is usually ~ten times faster than realtime for the "Deutsches Museum" example in Cartographer ROS.

If the compiler does not use vector math for the target platform, it will be slower by a factor.
We implement the math with Eigen, so the compiler can choose with operations to use for your target platform.

lukehutch, 2019-04-11:
Thanks. On the second part of my question, is there a hardware assembly and hardware/software integration getting started guide somewhere?

MichaelGrupp, 2020-10-16:
> Thanks. On the second part of my question, is there a hardware assembly and hardware/software integration getting started guide somewhere?

Not that I know of. Closing as answered.

closed 2020-10-16 by MichaelGrupp