cartographer-project/cartographer_ros #1259: Failed to process package 'cartographer_ros_msgs'

Karltlbm, 2019-04-10:
Hello

I am installing cartograpgher ROS on fedora 29.
I follow this tutorial (https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html).

$ catkin_make_isolated --install --use-ninja

```
Base path: /home/karlmontalban/Dev/rapidash/Testbench/src/tools/lab/carto/catkwin_ws
Source space: /home/karlmontalban/Dev/rapidash/Testbench/src/tools/lab/carto/catkwin_ws/src
Build space: /home/karlmontalban/Dev/rapidash/Testbench/src/tools/lab/carto/catkwin_ws/build_isolated
Devel space: /home/karlmontalban/Dev/rapidash/Testbench/src/tools/lab/carto/catkwin_ws/devel_isolated
Install space: /home/karlmontalban/Dev/rapidash/Testbench/src/tools/lab/carto/catkwin_ws/install_isolated
WARNING: Package name "ceres-solver" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 5 packages in topological order:
~~  - cartographer_ros_msgs
~~  - ceres-solver (plain cmake)
~~  - cartographer (plain cmake)
~~  - cartographer_ros
~~  - cartographer_rviz
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

==> Processing catkin package: 'cartographer_ros_msgs'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/karlmontalban/Dev/rapidash/Testbench/src/tools/lab/carto/catkwin_ws/build_isolated/cartographer_ros_msgs'
ninja: no work to do.
==> ninja -j12 -l12 in '/home/karlmontalban/Dev/rapidash/Testbench/src/tools/lab/carto/catkwin_ws/build_isolated/cartographer_ros_msgs'
[13/13] cd /home/karlmontalban/Dev/rapidash/Testbench/src/tools/lab/carto/catkwin_ws/build_isolated/cartographer_ros_msgs && /home/karlmont...me/karlmontalban/Dev/rapidash/Testbench/src/tools/lab/carto/catkwin_ws/src/cartographer_ros/cartographer_ros_msgs/msg/TrajectoryOptions.msg
==> ninja install in '/home/karlmontalban/Dev/rapidash/Testbench/src/tools/lab/carto/catkwin_ws/build_isolated/cartographer_ros_msgs'
[13/14] Install the project...
```

...... up to date .... 

ERROR : 

```
Unhandled exception of type 'RuntimeError':
Traceback (most recent call last):
  File "/opt/openrobots/lib/python2.7/site-packages/catkin/builder.py", line 965, in build_workspace_isolated
    number=index + 1, of=len(ordered_packages)
  File "/opt/openrobots/lib/python2.7/site-packages/catkin/builder.py", line 670, in build_package
    "'\n  This sometimes occurs when a non-catkin package is "
RuntimeError: No env.sh file generated at: '/home/karlmontalban/Dev/rapidash/Testbench/src/tools/lab/carto/catkwin_ws/install_isolated/env.sh'
  This sometimes occurs when a non-catkin package is interpreted as a catkin package.
  This can also occur when the cmake cache is stale, try --force-cmake.
<== Failed to process package 'cartographer_ros_msgs': 
  No env.sh file generated at: '/home/karlmontalban/Dev/rapidash/Testbench/src/tools/lab/carto/catkwin_ws/install_isolated/env.sh'
  This sometimes occurs when a non-catkin package is interpreted as a catkin package.
  This can also occur when the cmake cache is stale, try --force-cmake.
Command failed, exiting.
```

closed 2019-04-24 by Karltlbm

SimonDahrs, 2019-08-26:
@Karltlbm I am currently having the same issue. Did you manage to solve it, if so, how?

LOKAKSHAI, 2021-06-08:
I'm also getting the same error

Failed to process package 'cartographer_ros_msgs'

 Command '['/home/catkin_ws/devel_isolated/cartographer/env.sh', 'cmake', '/home/catkin_ws/src/cartographer_ros/cartographer_ros_msgs', '-DCATKIN_DEVEL_PREFIX=/home/catkin_ws/devel_isolated/cartographer_ros_msgs', '-DCMAKE_INSTALL_PREFIX=/home/catkin_ws/install_isolated', '-G', 'Unix Makefiles']' returned non-zero exit status 1