cartographer-project/cartographer_ros #1265: std::out_of_range error in running demo cartographer_paper_deutsches_museum.bag

UditSinghParihar, 2019-04-16:
1. Cartograhper on demo file:-
	1. $ roslaunch cartographer_ros demo_backpack_2d.launch bag_filename:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag

	2. Error:
	[ INFO] [1555409027.236835694, 1405076296.503967961]: I0416 15:33:47.000000 17593 ordered_multi_queue
.cc:172] All sensor data for trajectory 0 is available starting at '635406730964746083'.
terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at

        3. Distro- Ubuntu16.04, Kinetic. My installation process was error free. So what could be the possible reason.

Thanks.

pifon2a, 2019-04-17:
Are you using Cartographer provided with ROS?

UditSinghParihar, 2019-04-17:
Yeah I am using Cartographer with ROS, with instructions from this [page](https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html)

UditSinghParihar, 2019-04-18:
Ok, I solved it. I was basically using `PCL1.9`, which can only be used with `c++14`. So I change 
`CMakeLists.txt` in `cartographer_ros` to be compiled with `c++14`. This was causing the error.  After using `PCL1.7` which does not require to be used with `c++14`, I was able to build cartographer without changing anything in the `CMakeLists.txt`.  
I think we could close this issue.  

Thanks

closed 2019-04-18 by UditSinghParihar