cartographer-project/cartographer_ros #1266: Node(cartographer_occupancy_grid_node) get killed when subscibing to /map topic.

UditSinghParihar, 2019-04-16:
1. I am running my bag file with [this launch file](https://github.com/robotics-research-center/Loop_closure/blob/master/plot_slam/cartographer_scripts/my_backpack_2d.launch) and [this configuration file](https://github.com/robotics-research-center/Loop_closure/blob/master/plot_slam/cartographer_scripts/my_backpack_2d.lua). It works perfectly fine, but as soon as I subscribe to `/map` topic or try to visualize in 'rviz'. Then it is throwing the following error:

` exit code -11, cmd /home/cair/softwares/cartographer_ws/install_isolated/lib/cartographer_ros/cartographer_occupancy_grid_node -resolution 0.05 __name:=cartographer_occupancy_grid_node
`
2. Here is my [output](https://github.com/robotics-research-center/Loop_closure/blob/master/plot_slam/cartographer_scripts/rosbag_validate_output.txt)  of the `cartographer_rosbag_validate` on my bag_file, which have these topics:
```
/scan
/points
/imu
/odom
/tf
```

MichaelGrupp, 2019-04-16:
Exit code 11 is a segfault.

You can obtain a core dump to check the backtrace using GDB: https://jvns.ca/blog/2018/04/28/debugging-a-segfault-on-linux/

UditSinghParihar, 2019-04-16:
1. The output from gdb using backtrace:
`(gdb) backtrace 
#0  0x00000000004d13ec in boost::detail::function::void_function_obj_invoker1<cartographer_ros::(anonymous namespace)::Node::Node(double, double)::{lambda(boost::shared_ptr<cartographer_ros_msgs::SubmapList_<std::allocator<void> > const> const&)#1}, void, boost::shared_ptr<cartographer_ros_msgs::SubmapList_<std::allocator<void> > const> const&>::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<cartographer_ros_msgs::SubmapList_<std::allocator<void> > const> const&) ()
#1  0x00000000004d4d2e in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<cartographer_ros_msgs::SubmapList_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<cartographer_ros_msgs::SubmapList_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<cartographer_ros_msgs::SubmapList_<std::allocator<void> > const>) ()
#2  0x00000000004d80a2 in ros::SubscriptionCallbackHelperT<boost::shared_ptr<cartographer_ros_msgs::SubmapList_<std::allocator<void> > const> const&, void>::call(ros::SubscriptionCallbackHelperCallParams&) ()
#3  0x00007f5508ca1d8d in ?? ()
#4  0x0000000025180f3e in ?? ()
#5  0x00007ffc0b144380 in ?? ()
#6  0x00000000022c6d90 in ?? ()
#7  0x00000000022c6d90 in ?? ()
#8  0x00000000022c7420 in ?? ()
#9  0x00000000022c7400 in ?? ()
#10 0x0000000000000000 in ?? ()`

2. The output from valgrind on `cartographer_occupancy_grid_node`:
`==1816== Process terminating with default action of signal 11 (SIGSEGV): dumping core
==1816==  General Protection Fault
==1816==    at 0x4D13EC: boost::detail::function::void_function_obj_invoker1<cartographer_ros::(anonymous namespace)::Node::Node(double, double)::{lambda(boost::shared_ptr<cartographer_ros_msgs::SubmapList_<std::allocator<void> > const> const&)#1}, void, boost::shared_ptr<cartographer_ros_msgs::SubmapList_<std::allocator<void> > const> const&>::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<cartographer_ros_msgs::SubmapList_<std::allocator<void> > const> const&) (in /home/cair/softwares/cartographer_ws/install_isolated/lib/cartographer_ros/cartographer_occupancy_grid_node)
==1816==    by 0x4D4D2D: boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<cartographer_ros_msgs::SubmapList_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<cartographer_ros_msgs::SubmapList_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<cartographer_ros_msgs::SubmapList_<std::allocator<void> > const>) (in /home/cair/softwares/cartographer_ws/install_isolated/lib/cartographer_ros/cartographer_occupancy_grid_node)
==1816==    by 0x4D80A1: ros::SubscriptionCallbackHelperT<boost::shared_ptr<cartographer_ros_msgs::SubmapList_<std::allocator<void> > const> const&, void>::call(ros::SubscriptionCallbackHelperCallParams&) (in /home/cair/softwares/cartographer_ws/install_isolated/lib/cartographer_ros/cartographer_occupancy_grid_node)
==1816==    by 0x5FE4D8C: ros::SubscriptionQueue::call() (in /opt/ros/kinetic/lib/libroscpp.so)
==1816==    by 0x5F8A837: ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) (in /opt/ros/kinetic/lib/libroscpp.so)
==1816==    by 0x5F8C23A: ros::CallbackQueue::callAvailable(ros::WallDuration) (in /opt/ros/kinetic/lib/libroscpp.so)
==1816==    by 0x5FE8E38: ros::SingleThreadedSpinner::spin(ros::CallbackQueue*) (in /opt/ros/kinetic/lib/libroscpp.so)
==1816==    by 0x5FCDE9A: ros::spin() (in /opt/ros/kinetic/lib/libroscpp.so)
==1816==    by 0x4CEDFE: main (in /home/cair/softwares/cartographer_ws/install_isolated/lib/cartographer_ros/cartographer_occupancy_grid_node)
--1816-- Discarding syms at 0xe9db1b0-0xe9e12a1 in /lib/x86_64-linux-gnu/libnss_files-2.23.so due to munmap()
==1816== 
==1816== HEAP SUMMARY:
==1816==     in use at exit: 2,663,988 bytes in 1,065 blocks
==1816==   total heap usage: 7,044 allocs, 5,979 frees, 14,946,358 bytes allocated
==1816== 
==1816== Searching for pointers to 1,065 not-freed blocks
==1816== Checked 45,370,192 bytes
==1816== 
==1816== LEAK SUMMARY:
==1816==    definitely lost: 0 bytes in 0 blocks
==1816==    indirectly lost: 0 bytes in 0 blocks
==1816==      possibly lost: 1,840 bytes in 5 blocks
==1816==    still reachable: 2,662,148 bytes in 1,060 blocks
==1816==         suppressed: 0 bytes in 0 blocks
==1816== Rerun with --leak-check=full to see details of leaked memory
==1816== 
==1816== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
==1816== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
Killed`

3. It looks like some boost function is causing segmentation fault, but I am not able to figure it out, or is it due to some wrong parameters in my configuration file?

Thanks

MichaelGrupp, 2019-04-17:
Looks like the error happens in a ROS callback function in the occupancy grid node. Can't say more at the moment.

UditSinghParihar, 2019-04-18:
Ok, I solved it. I was basically using `PCL1.9`, which can only be used with `c++14`. So I change 
`CMakeLists.txt` in `cartographer_ros` to be compiled with `c++14`. This was causing the error.  After using `PCL1.7` which does not require to be used with `c++14`, I was able to build cartographer without changing anything in the `CMakeLists.txt`.  
I think we could close this issue.  

Thanks

closed 2019-04-18 by UditSinghParihar