cartographer-project/cartographer_ros #1272: Tuning help for home-made robot with home-made LIDAR

samialperen, 2019-04-24:
Hello Guys,

I am trying to improve cartographer result for my home-made LIDAR. 
[Here](https://answers.ros.org/question/321676/tuning-gmapping-parameters-or-alternative-slam-algorithm/?comment=321803#post-id-321803) is a link that I recently asked in roswiki.

1. rosbag validate result:
`E0424 20:04:08.134719 25812 rosbag_validate_main.cc:350] frame_id "odom" on topic /odom has serialization time 1555961461.908083810 but sensor time 1555953734.416547069 differing by -7727.49 s.
E0424 20:04:08.135022 25812 rosbag_validate_main.cc:350] frame_id "odom" on topic /odom has serialization time 1555961461.952052829 but sensor time 1555953734.466559309 differing by -7727.49 s.
E0424 20:04:08.135043 25812 rosbag_validate_main.cc:350] frame_id "odom" on topic /odom has serialization time 1555961462.001106965 but sensor time 1555953734.516543580 differing by -7727.48 s.
E0424 20:04:08.135164 25812 rosbag_validate_main.cc:131] Input contains transform message from frame_id "map" to child_frame_id "odom". This is almost always output published by cartographer and should not appear as input. (Unless you have some complex remove_frames configuration, this is will not work. Simplest solution is to record input without cartographer running.)
W0424 20:04:08.141655 25812 rosbag_validate_main.cc:166] Sensor with frame_id "base_scan" measurements overlap in time. Previous range message, ending at time stamp 636915505605857072, must finish before current range message, which ranges from 636915505596746061 to 636915505606589811
W0424 20:04:08.141710 25812 rosbag_validate_main.cc:166] Sensor with frame_id "base_scan" measurements overlap in time. Previous range message, ending at time stamp 636915505606589811, must finish before current range message, which ranges from 636915505597370137 to 636915505607213887
W0424 20:04:08.141748 25812 rosbag_validate_main.cc:166] Sensor with frame_id "base_scan" measurements overlap in time. Previous range message, ending at time stamp 636915505607213887, must finish before current range message, which ranges from 636915505598114108 to 636915505607957858
W0424 20:04:08.193042 25812 rosbag_validate_main.cc:203] Sensor with frame_id "base_scan" range measurements have longest overlap of 0.921967 s
E0424 20:04:08.193071 25812 rosbag_validate_main.cc:382] Point data (frame_id: "base_scan") has a large gap, largest is 0.875337 s, recommended is [0.0005, 0.05] s with no jitter.
I0424 20:04:08.193133 25812 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/scan" (frame_id: "base_scan"):
Count: 228  Min: 0.062408  Max: 0.875337  Mean: 0.745079
[0.062408, 0.143701)	                   #	Count: 17 (7.456141%)	Total: 17 (7.456141%)
[0.143701, 0.224994)	                    	Count: 0 (0.000000%)	Total: 17 (7.456141%)
[0.224994, 0.306287)	                    	Count: 0 (0.000000%)	Total: 17 (7.456141%)
[0.306287, 0.387579)	                    	Count: 1 (0.438596%)	Total: 18 (7.894737%)
[0.387579, 0.468872)	                    	Count: 0 (0.000000%)	Total: 18 (7.894737%)
[0.468872, 0.550165)	                    	Count: 0 (0.000000%)	Total: 18 (7.894737%)
[0.550165, 0.631458)	                    	Count: 0 (0.000000%)	Total: 18 (7.894737%)
[0.631458, 0.712751)	                    	Count: 0 (0.000000%)	Total: 18 (7.894737%)
[0.712751, 0.794044)	              ######	Count: 70 (30.701754%)	Total: 88 (38.596493%)
[0.794044, 0.875337]	        ############	Count: 140 (61.403507%)	Total: 228 (100.000000%)
I0424 20:04:08.193343 25812 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/odom" (frame_id: "odom"):
Count: 4470  Min: 0.044489  Max: 0.055512  Mean: 0.050000
[0.044489, 0.045592)	                    	Count: 2 (0.044743%)	Total: 2 (0.044743%)
[0.045592, 0.046694)	                    	Count: 1 (0.022371%)	Total: 3 (0.067114%)
[0.046694, 0.047796)	                    	Count: 1 (0.022371%)	Total: 4 (0.089485%)
[0.047796, 0.048898)	                    	Count: 1 (0.022371%)	Total: 5 (0.111857%)
[0.048898, 0.050001)	         ###########	Count: 2421 (54.161076%)	Total: 2426 (54.272930%)
[0.050001, 0.051103)	           #########	Count: 2039 (45.615211%)	Total: 4465 (99.888145%)
[0.051103, 0.052205)	                    	Count: 1 (0.022371%)	Total: 4466 (99.910515%)
[0.052205, 0.053308)	                    	Count: 1 (0.022371%)	Total: 4467 (99.932884%)
[0.053308, 0.054410)	                    	Count: 1 (0.022371%)	Total: 4468 (99.955254%)
[0.054410, 0.055512]	                    	Count: 2 (0.044743%)	Total: 4470 (100.000000%)
`
2. My launch file:
`<launch>
  <param name="/use_sim_time" value="true" />

  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename kamu_robotu.lua"
      output="screen">
  </node>

  <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
      type="cartographer_occupancy_grid_node" args="-resolution 0.05" />
</launch>`
My Lua file:
`include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 1.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true

TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.linear_search_window = 0.1
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.angular_search_window = math.rad(35.)

-- Parameters added later
--TRAJECTORY_BUILDER_2D.min_range = 0.001
TRAJECTORY_BUILDER_2D.max_range = 1.0

POSE_GRAPH.optimization_problem.huber_scale = 1e2

return options`
3. Two bag files that I use:
[real_data1.bag](https://uploadfiles.io/69w733je) 64 point  1 Hz LIDAR
[real_data2.bag](https://ufile.io/7qjao693) 64 point 4 Hz LIDAR (I increased rotation speed of the LIDAR)

4. [Here](https://ibb.co/5KmGQVs) is the link for the cartographer result for real_data1.bag

I really appreciate for your help!

samialperen, 2019-04-24:
@MichaelGrupp  @esteve @SirVer @damienrg @damonkohler

clement2449, 2019-05-14:
hi:) is there any issues here? i have the same problem

samialperen, 2019-05-14:
@clement2449, no improvement unfortunately :(

clement2449, 2019-05-14:
@samialperen do you want my files maybe it can help

clement2449, 2019-05-14:
[image]

clement2449, 2019-05-14:
oh i forgot sorry dont put the sim time to true if you have a hardware device. i did that before and i didn't work

MichaelGrupp, 2020-10-27:
While we can't help much with debugging this custom setup, I recommend to read the output of the validate tool. There are multiple issues with timestamps, TF etc. in your data.

closed 2020-10-27 by MichaelGrupp