cartographer-project/cartographer_ros #1275: map rotates

Colidar, 2019-04-25:
Hi all,

i have a problem using realtime 2D SLAM with an rplidar a2 during mapping process. When i'm rotating the lidar the new submaps are also rotating and build something like a second map.

[image]

I am not using odometrie.
Does anyone know if this is normal or i have to change something in my files? 
My files are on dropbox:
https://www.dropbox.com/sh/yutejspswzqqk8o/AACX8RzWfwWcmFipI0zXYRRQa?dl=0

MrGitGo, 2019-04-29:
I'm not a big help but what I see is that you can change the tracking frame from laser to base_link and provide odom frame to true. And you could look here in the issue page to find someone who has the same setup and tune your cartographer like them.

gaschler, 2019-04-30:
Please post the *text* output of cartographer_rosbag_validate.
A common reason is gaps in laser or imu data.

jotadepicas, 2020-01-14:
Hi, I'm experiencing a pretty similar issue, my map also rotates, maybe not as drastically as the one posted here, but definitely there are some rotation issues. Below I'll paste the validate output. What I want to check is if my understanding of the detected issues is correct. (it would be nice also to know what happened to @Colidar issue). 

So, `cartographer_rosbag_validate` detects a couple of issues with my LIDAR sensor data, but I'm not sure exactly how to interpret them. Basically, validate detects two issues for /scan messages:

### There's a large time difference (sometimes a negative one) between serialization time and sensor time

Having a large time difference between makes me think about processing power, but could it also be networking? Is this serialization time taking into account the time it takes for the msg to reach roscore or something like that? In particular, how can this difference be negative? Could this indicate a clock sync issue?

I've seen this issue is more or less ignored by people doing just offline mapping reconstruction, but I'm doing SLAM, so I think this could be an actual issue in my scenario.

### There are gaps in between /scan messages that are above recommended values. 

I'm using YDLIDAR X4, which states it works at 7 hz. My understanding from this is that it will produce no more than 7 /scan messages per second, so the gap between them will be aprox 142 ms. Validation says "`recommended is [0.0005, 0.05] s with no jitter`". How can I have such small gaps, if my sensor works at 7 hz? I've seen videos of people getting pretty good results with YDLIDAR X4, so I'm guessing there's a misconception on my side about this, otherwise YDLIDAR woudn't even work for anyone :thinking: .

This is the text output from `cartographer_rosbag_validate`:

```
W0110 15:29:44.522763 19183 rosbag_validate_main.cc:352] frame_id "laser_frame" on topic /scan has serialization time 1578680789.040276050 but sensor time 1578680788.747516000 differing by -0.29276 s.
E0110 15:29:44.525462 19183 rosbag_validate_main.cc:350] frame_id "laser_frame" on topic /scan has serialization time 1578680797.428982973 but sensor time 1578680796.891624000 differing by -0.537359 s.
E0110 15:29:44.525600 19183 rosbag_validate_main.cc:350] frame_id "laser_frame" on topic /scan has serialization time 1578680797.570624113 but sensor time 1578680797.007191000 differing by -0.563433 s.
E0110 15:29:44.525665 19183 rosbag_validate_main.cc:350] frame_id "laser_frame" on topic /scan has serialization time 1578680797.713319063 but sensor time 1578680797.120317000 differing by -0.593002 s.
E0110 15:29:44.537004 19183 rosbag_validate_main.cc:382] Point data (frame_id: "laser_frame" has a large gap, largest is 0.125339 s, recommended is [0.0005, 0.05] s with no jitter.
I0110 15:29:44.537109 19183 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/scan" (frame_id: "laser_frame":
Count: 392  Min: 0.102170  Max: 0.125339  Mean: 0.114846
[0.102170, 0.104487)	                   	Count: 2 (0.510204%) Total: 2 (0.510204%)
[0.104487, 0.106804)	                   	Count: 7 (1.785714%) Total: 9 (2.295918%)
[0.106804, 0.109121)	                 ##	Count: 31 (7.908163%)	Total: 40 (10.204082%)
[0.109121, 0.111438)	                   	Count: 0 (0.000000%) Total: 40 (10.204082%)
[0.111438, 0.113755)	                  #	Count: 26 (6.632653%)	Total: 66 (16.836735%)
[0.113755, 0.116071)	     ##############	Count: 273 (69.642860%)	Total: 339 (86.479591%)
[0.116071, 0.118388)	                  #	Count: 13 (3.316327%)	Total: 352 (89.795921%)
[0.118388, 0.120705)	                   	Count: 2 (0.510204%) Total: 354 (90.306122%)
[0.120705, 0.123022)	                  #	Count: 26 (6.632653%)	Total: 380 (96.938774%)
[0.123022, 0.125339]	                  #	Count: 12 (3.061224%)	Total: 392 (100.000000%)
```

MichaelGrupp, 2020-10-26:
While scan matching might be improved by parameter tuning, be aware that strong rotations can be suboptimal if no complementary high-frequency sensor like an IMU is used. Closing as we are not able to provide more support for your setups.

closed 2020-10-26 by MichaelGrupp