cartographer-project/cartographer_ros #1277: set protobuf3 for cartographer_ros and there still exist protobuf2

crankler, 2019-05-05:
I need protoc 2.6.1 for some ros packages, but now when I install cartigrapher on my computer,
there say I need to install protoc3, so I use this way to install protoc3.7 in /usr/protobuf
https://blog.csdn.net/qq_39024800/article/details/82751323

And I also modify the CMakeLists.txt in cartographer as follow:
SET(PROTOBUF_DIR /usr/protobuf)
find_package(Protobuf 3 REQUIRED)

but when I compile use command:catkin_make_isolated --install --use-ninja
there still cannot find proto3.

So if I want to use protoc3 for cartographer and protoc2 for other ros package,
how can I set in cartographer package?
Thank you!!!

gaschler, 2019-05-16:
You don't need to install protobuf in the system to use it in Cartographer.
One way to edit the cmake settings is using `ccmake` in the build folder.
Once you know all the settings to change, you can later specify `-D...=...` arguments to catkin that are passed to cmake.
The tricky thing is when having multiple protobuf versions,
you must use the exact matching versions of protobuf compiler, header files and libraries and not mix versions.

crankler, 2019-05-22:
Thank you for your reply!
So can you tell me the detail setting on this package!

ojura, 2019-05-22:
@gaschler a much better alternative is to set `CMAKE_PREFIX_PATH` to the install location of Protobuf 3 and let `find_package(Protobuf)` set up everything, instead of manually setting all CMake variables using `-DPROTOBUF_..`.

MichaelGrupp, 2020-10-27:
@ojura's approach looks good. On newer Ubuntu versions, I would recommend to use just the version that is installed via rosdep. The build documentation was also changed accordingly.

closed 2020-10-27 by MichaelGrupp