cartographer-project/cartographer_ros #1279: when using offline_node and asserts_writer, i have to keep bag_filenames in time sequence, right?

qmx066, 2019-05-08:
hi team,

these days i found an interesting thing, when i ran offline node with several bag files, i can get pbstream file well, then ran asserts_writer to get some more output data, i got problem that the first trajectory in pbstream was assigned with the second package's timestamp. such as,

rosrun cartographer_ros offline_node   -bag_filenames:=newer_bag.bag,older_bag.bag;
rosrun cartographer_ros asserts_writer -bag_filenames:=newer_bag.bag,older_bag.bag -state_filename:=newer_bag.bag.pbstream

then asserts_writer failed.

i checked timestamp in pbstream, the second(the older one) bag's timestamp was loaded first, so it failed.

my QUESTION is, is it designed so? do i have to put bag files in argument list following time line?

BR,

ojura, 2019-05-08:
Ah yes, unfortunately we did not get around yet to use the bag multiplexer in the assets writer, so it is sensitive to the order of specifying bags.

qmx066, 2019-05-09:
> Ah yes, unfortunately we did not get around yet to use the bag multiplexer in the assets writer, so it is sensitive to the order of specifying bags.

thanks for the confirmation

a further question, how about bags containing time overlap? is asserts writer handle this?

ojura, 2019-05-09:
Not sure, I think not. Supporting simultaneous bags was precisely the reason for introducing the bag multiplexer. A PR for using it in the offline node would be welcome.

closed 2020-10-27 by MichaelGrupp