cartographer-project/cartographer_ros #1280: pbstrem to yaml

ytgcljj, 2019-05-10:
nuc16@nuc16:/opt/openresty/nginx/scripts$ rosrun cartographer_ros cartographer_pbstream_to_ros_map -map_filestem=/home/nuc16/catkin_ws/install_isolated/share/cartographer_ros/maps/ros -pbstream_filename=/home/nuc16/catkin_ws/install_isolated/share/cartographer_ros/maps/ros.pbstream
I0510 17:39:21.586185  9304 pbstream_to_ros_map_main.cc:52] Loading submap slices from serialized data.
I0510 17:39:21.726846  9304 pbstream_to_ros_map_main.cc:75] Generating combined map image from submap slices.

gaschler, 2019-05-14:
Closing as not a question/not actionable.

closed 2019-05-14 by gaschler

ytgcljj, 2019-05-14:
rosrun cartographer_ros cartographer_pbstream_to_ros_map -pbstream_filename=/home/nuc16/catkin_ws/install_isolated/share/cartographer_ros/maps/ros.pbstream -map_filestem=/home/nuc16/ros
I0514 18:11:03.544162 14931 pbstream_to_ros_map_main.cc:52] Loading submap slices from serialized data.
段错误

ytgcljj, 2019-05-16:
nuc16@nuc16:/opt/openresty/nginx/html/ros$ rosrun cartographer_ros cartographer_pbstream_to_ros_map -pbstream_filename=/home/nuc16/catkin_ws/install_isolated/share/cartographer_ros/maps/ros.pbstream -map_filestem=/opt/openresty/nginx/html/ros/ros
I0516 11:02:17.867508 18376 pbstream_to_ros_map_main.cc:52] Loading submap slices from serialized data.
段错误