cartographer-project/cartographer_ros #1283: How to evaluate the landmark-fusion error?

mojovski, 2019-05-11:
Dear all,
I stumbled again over the landmark integration issue. As far as I can see, these are received by cartographer but the correction of the global pose is not taking place.

### The scene:
1) The map is generated and stored as a pbstream file (system is shut off)
2) The map from (1) is loaded, the robot goes to a working station where a landmark is located. It "remembers" it (i store it inside redis) and drives to somewhere else. The system is shut off.
3) The map from (1) is loaded, when the robot comes back to the landmark, the error between the remembered landmark (I store them inside redis) on the map frame and the currently observed landmark on the map frame is about 0.4m.

From my understanding the error is quite high for the "normal" slam. Also, it should have been corrected by the landmark integration.

### The strange things

* The local costmap is overlapping with the global map and by visual inspection I would not say, that the error is 40cm

* With increased weight on the landmarks, cartographer fails to perform loop-closing. I guess, this is due to the relation of the loop closing metrics and the landmark metrics. Does high landmark weights reduce the probability of a loop closing being accepted by the constraint builder? Thus, the weight of the landmark is relative to POSE_GRAPH.constraint_builder.loop_closure_translation_weight ?

* When the landmark is observed in step (3) and published to cartographer, is cartographer required to have this landmark inside the loaded *.pbstream map in step (1) in order to use it for correction?

* If I set the rotation weight to 1 but the translation weight to 1e22, the rotation error of the slam is becoming also quite high (loop closing also not working anymore)

 * How does the landmark fusion relate to POSE_GRAPH.constraint_builder.sampling_ratio?

These are quite many questions. 
If you feel you have a deterministic  feeling about one of them, I would love to read your opinion.

Thanks alot!

gaschler, 2019-05-16:
I don't fully understand what exactly you are doing, but I try to answer some questions.

>  Does high landmark weights reduce the probability of a loop closing being accepted by the constraint builder?

Sure, when those landmark poses are worse or presented in a wrong coordinate frame.
Then, pose graph is pushed in the wrong direction, latest pose estimate is worse and loop closure searches in the wrong area.

> How does the landmark fusion relate to POSE_GRAPH.constraint_builder.sampling_ratio?

This parameter defines what fraction of pairs of scans and submaps are searched for loop closures. No more, no less.

Hope this helps

mojovski, 2019-05-17:
Hi @gaschler , thank you for your comments.

> Does high landmark weights reduce the probability of a loop closing being accepted by the constraint builder?
Maybe I describe the observation in more detail:
* The robot starts at (0,0; 0) (x,y,phi) and observes a landmark. This observation is always the same .
* Now, when I feed cartographer with the (presumably) correct landmark observation, cartographer fails to detect and to close loop closures.

Anyway, I feel that I will have to debug it on my own first to get a better picture.

MichaelGrupp, 2020-10-27:
Closing for inactivity, we can't invest time to help you with your setup anymore unfortunately.

closed 2020-10-27 by MichaelGrupp