cartographer-project/cartographer_ros #1285: How to fix "Point data (lidar) has a large gap"

SyrineZ, 2019-05-13:
Hi,

I'm trying to run cartographer for my robot and the cartographer_rosbag_validate tells me that "Point data (frame_id: "rplidar") has a large gap, largest is 0.201 s, recommended is [0.0005, 0.05] s with no jitter".
Here's my .lua file

include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false,
  use_pose_extrapolator = true,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 10,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.num_accumulated_range_data = 1

return options

and these are the parameters of the lidar

<gazebo reference="rplidar">
    <material>Gazebo/light_black</material>
    <sensor type="ray" name="rplidar_A1M8">
      <pose>0 0 0 0 0 0</pose>
      <visualize>true</visualize>
      <update_rate>5.5</update_rate>
      <ray>
        <scan>
          <horizontal>
            <samples>1450</samples>
            <resolution>1</resolution>
            <min_angle>0.0</min_angle>
            <max_angle>6.28319</max_angle>
          </horizontal>
        </scan>
        <range>
          <min>0.15</min>
          <max>12</max>
          <resolution>0.005</resolution>
        </range>
     <noise>
      <type>gaussian</type>
      <mean>0.0</mean>
      <stddev>0.1</stddev>
    </noise>
      </ray>
      <plugin name="rplidar_controller" filename="libgazebo_ros_laser.so">
        <topicName>scan</topicName>
        <frameName>rplidar</frameName>
      </plugin>
    </sensor>
</gazebo>

SyrineZ, 2019-05-13:
<gazebo reference="rplidar">
    <material>Gazebo/light_black</material>
    <sensor type="ray" name="rplidar_A1M8">
      <pose>0 0 0 0 0 0</pose>
      <visualize>true</visualize>
      <update_rate>5.5</update_rate>
      <ray>
        <scan>
          <horizontal>
            <samples>1450</samples>
            <resolution>1</resolution>
            <min_angle>0.0</min_angle>
            <max_angle>6.28319</max_angle>
          </horizontal>
        </scan>
        <range>
          <min>0.15</min>
          <max>12</max>
          <resolution>0.005</resolution>
        </range>
     <noise>
      <type>gaussian</type>
      <mean>0.0</mean>
      <stddev>0.1</stddev>
    </noise>
      </ray>
      <plugin name="rplidar_controller" filename="libgazebo_ros_laser.so">
        <topicName>scan</topicName>
        <frameName>rplidar</frameName>
      </plugin>
    </sensor>
</gazebo>

gaschler, 2019-05-16:
> use_pose_extrapolator = true,

Remove this line. Apparently you use an old version of cartographer_ros that does not yet know this flag.

Closing as answered.

closed 2019-05-16 by gaschler