cartographer-project/cartographer_ros #1286: Advanced tuning advices, VLP32 + IMU

Karltlbm, 2019-05-14:
Hi,

I managed to run Cartographer_ros on my setup (velodyne lidar + imu for indoor mapping).
I'm having difficulties to understand how to properly tune all the parameters, I believe cartographer runs pretty well but that the final result could be much cleaner.
My goal is to do 3D reconstruction, transform the final pointcloud into meshes so the pointcloud has to be as clean as possible. But as it is now, it is hardly achievable to transform it to a mesh.

Could you tell me what you think ?

You can access to my bag file here:
https://drive.google.com/open?id=110mzgX5tOJuvymWZ9KPcUoN_NYViJNCt

My ply final pointcloud file (you can see it in CloudCompare):
https://drive.google.com/open?id=1lCEdH1jpydoqiViFVNmJIV53SpdYt3xY

It is created by the simplest assets writer lua file:

```
options = {
  tracking_frame = "imu_link",
  pipeline = {
    {
      action = "min_max_range_filter",
      min_range = 2.5,
      max_range = 10,
    },
    {
      action = "dump_num_points",
    },
    {
      action = "write_ply",
      filename = "points.ply",
    },
  }
}
```
return options

Here is my lua config file:
[my_config_lua.txt](https://github.com/googlecartographer/cartographer_ros/files/3176514/my_config_lua.txt)

Here is a screen of the final map with the trajectory:
[image]

And here is a screen of the pointcloud (filtered by a random sampling 20M points initially to 5M):
[image]

gaschler, 2019-05-14:
Please post the text output of cartographer_rosbag_validate.
Input data is almost never perfect and we can easily help you there.

> final result could be much cleaner.

Agree. The map looks fuzzy. 

>  i can't reach to any of the submaps created and I don't know why...

Follow the tutorial and first verify you can run rviz with the cartographer_rviz plugin.
Most common mistake there is not running `source setup.bash`

I took a quick look at your lua file. It's hard to see what non-default parameters you have.

>  num_point_clouds = 1,
> TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 2

You have only one lidar, so set this to one is one rotation sends one point cloud message.

> TRAJECTORY_BUILDER_3D.ceres_scan_matcher.occupied_space_weight_0 = 1 -- HIGH RESOLUTION POINT CLOUD
> TRAJECTORY_BUILDER_3D.ceres_scan_matcher.occupied_space_weight_1 = 6 -- LOW RESOLUTION POINT CLOUD

Don't change these parameters, this doesn't do good.

> TRAJECTORY_BUILDER_3D.use_online_correlative_scan_matching = true

Why?

1 VLP32 is very close to our default config, you should only need to tune 1-2 knobs and then it will be good.

Hope this helps.

Karltlbm, 2019-05-14:
Hi, thank you very much for your help ! Here is the output of rosbag validate:
```
W0514 13:13:52.926021  4854 rosbag_validate_main.cc:353] frame_id "vlp_link" on topic VLP has serialization time 1555685277.224022653 but sensor time 1555685277.122502354 differing by -0.10152 s.
I0514 13:13:54.907027  4854 rosbag_validate_main.cc:399] Time delta histogram for consecutive messages on topic "IMU" (frame_id: "imu_link"):
Count: 13331  Min: 0.00468984  Max: 0.0152779  Mean: 0.00999991
[0.004690, 0.005749)	                    	Count: 20 (0.150026%)	Total: 20 (0.150026%)
[0.005749, 0.006807)	                    	Count: 0 (0%)	Total: 20 (0.150026%)
[0.006807, 0.007866)	                    	Count: 0 (0%)	Total: 20 (0.150026%)
[0.007866, 0.008925)	                    	Count: 1 (0.00750131%)	Total: 21 (0.157528%)
[0.008925, 0.009984)	             #######	Count: 4699 (35.2487%)	Total: 4720 (35.4062%)
[0.009984, 0.011043)	       #############	Count: 8590 (64.4363%)	Total: 13310 (99.8425%)
[0.011043, 0.012101)	                    	Count: 1 (0.00750131%)	Total: 13311 (99.85%)
[0.012101, 0.013160)	                    	Count: 0 (0%)	Total: 13311 (99.85%)
[0.013160, 0.014219)	                    	Count: 15 (0.11252%)	Total: 13326 (99.9625%)
[0.014219, 0.015278]	                    	Count: 5 (0.0375066%)	Total: 13331 (100%)
E0514 13:13:54.907142  4854 rosbag_validate_main.cc:383] Point data (frame_id: "vlp_link") has a large gap, largest is 0.103537 s, recommended is [0.0005, 0.05] s with no jitter.
I0514 13:13:54.907320  4854 rosbag_validate_main.cc:399] Time delta histogram for consecutive messages on topic "VLP" (frame_id: "vlp_link"):
Count: 1332  Min: 0.0974328  Max: 0.103537  Mean: 0.0999996
[0.097433, 0.098043)	                    	Count: 5 (0.375375%)	Total: 5 (0.375375%)
[0.098043, 0.098654)	                   #	Count: 34 (2.55255%)	Total: 39 (2.92793%)
[0.098654, 0.099264)	                  ##	Count: 106 (7.95796%)	Total: 145 (10.8859%)
[0.099264, 0.099875)	             #######	Count: 436 (32.7327%)	Total: 581 (43.6186%)
[0.099875, 0.100485)	             #######	Count: 478 (35.8859%)	Total: 1059 (79.5045%)
[0.100485, 0.101095)	                 ###	Count: 192 (14.4144%)	Total: 1251 (93.9189%)
[0.101095, 0.101706)	                   #	Count: 62 (4.65465%)	Total: 1313 (98.5736%)
[0.101706, 0.102316)	                    	Count: 11 (0.825826%)	Total: 1324 (99.3994%)
[0.102316, 0.102927)	                    	Count: 6 (0.45045%)	Total: 1330 (99.8499%)
[0.102927, 0.103537]	                    	Count: 2 (0.15015%)	Total: 1332 (100%)
```

About the submaps, I do source everytime but I will try the rviz plugin...

I changed the parameters like you said: the result is that the process is way faster (because of the online correlative scan matcher off) but the final map is worse.

Here is a screenshot of the map: 
[image]

As my goal is 3D mesh reconstruction, I don't care so much about speed of processing but require more quality about the final pointcloud, should i keep the online scan matcher on ?

gaschler, 2019-05-14:
A general advice, tune for 2D first. 2D is easier to tune perfectly, and then you only change a few things to do 3D.

> I changed the parameters like you said: the result is that the process is way faster (because of the online correlative scan matcher off) but the final map is worse.

What do you mean by final map? rviz view of submaps? asset writer output? combined point cloud?

> W0514 13:13:52.926021  4854 rosbag_validate_main.cc:353] frame_id "vlp_link" on topic VLP has serialization time 1555685277.224022653 but sensor time 1555685277.122502354 differing by -0.10152 s.

If you run the asset writer, this is the cause because it would use wrong timestamps.

How sure are you your IMU to lidar transform is correct?

> As my goal is 3D mesh reconstruction, I don't care so much about speed of processing but require more quality about the final pointcloud, should i keep the online scan matcher on ?

No.
You could tune for more loop closure (inter) constraints. And use smaller voxels and filter less.

Karltlbm, 2019-05-14:
I will look more closely on how to tune the 2D processing and come back with my results later.

Do you know the origin and/or how I can correct this serialization time difference ?

Thank you very much for your help. I will keep updating this issue.

Karltlbm, 2019-05-15:
Hello again,

I am now running cartographer 2D using the default configuration files.

I only changed these two lines because the result was way better:
```
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.translation_weight = 3
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.rotation_weight = 60
```

I think that the result is still very improvable, it looks like this:
[image]

What do you think ?

Once again, I can't get the submaps but I opened another issue for this https://github.com/googlecartographer/cartographer_ros/issues/1287

Here is my lua config file:
[my_config3.txt](https://github.com/googlecartographer/cartographer_ros/files/3182387/my_config3.txt)

closed 2020-11-06 by MichaelGrupp