cartographer-project/cartographer_ros #1287: Submaps

Karltlbm, 2019-05-15:
I think this is a simple issue but i can't find any clear answer here...

I am using cartographer_ros. No matter about the bags or the config, I can recieve the map created by cartographer but I just can't reach the submaps created.

My launch is as default for the rviz node:
```
<node name="rviz" pkg="rviz" type="rviz" required="true"
      args="-d $(find cartographer_ros)/configuration_files/my_demo2d.rviz" />
```

But I recieve the following error:
`[ERROR] [1557918365.856351064, 1555685279.916522078]: PluginlibFactory: The plugin for class 'Submaps' failed to load.  Error: According to the loaded plugin descriptions the class Submaps with base class type rviz::Display does not exist. Declared types are  rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/WrenchStamped`

Is it an issue about my ROS version ?
Or the way I builded cartographer_ROS ?

What do you think ?

Screenshot of rviz:
[image]

gaschler, 2019-05-15:
This depends on your installation. Have you installed cartographer_ros or compiled it? Which version?
Make sure cartographer_rviz is built.
Make sure you are not mixing versions and source the right ROS setup scripts.

Karltlbm, 2019-05-17:
I compiled it, cartographer_rviz is built and i did source devel_isolated/setup.bash.

Still not working... It's not a big issue but still weird.

omair-khalid, 2019-06-12:
Try  'source install_isolated/setup.bash' in your catkin folder.

Ellon, 2019-07-01:
Hello @omair-khalid, I'm having the same problem as @Karltlbm. I cannot `source install_isolated/setup.bash` because `catkin_make_isolated --install --use-ninja` fails with the error: 

```
$ catkin_make_isolated --install --use-ninja -DCMAKE_CXX_FLAGS='-DEIGEN_MAX_ALIGN_BYTES=16' -DCMAKE_BUILD_TYPE=Release
(...)
RuntimeError: No env.sh file generated at: '/home/mellon/Sandbox/catkin_ws/install_isolated/env.sh'
  This sometimes occurs when a non-catkin package is interpreted as a catkin package.
  This can also occur when the cmake cache is stale, try --force-cmake.
<== Failed to process package 'cartographer_ros_msgs': 
  No env.sh file generated at: '/home/mellon/Sandbox/catkin_ws/install_isolated/env.sh'
  This sometimes occurs when a non-catkin package is interpreted as a catkin package.
  This can also occur when the cmake cache is stale, try --force-cmake.
Command failed, exiting.
```

Adding `--force-cmake` as indicated in the message error did not solve the problem.

I know that this is related to [ROS REP 136](http://www.ros.org/reps/rep-0136.html) as seen [here](http://answers.ros.org/question/231666/catkin_make_isolated-does-not-create-envsh/). I noticed that the compilation of `cartographer_ros_msgs` adds a file `${INSTALL_SPACE}/etc/ros/env.sh`, instead of `${INSTALL_SPACE}/env.sh` as expected.

Am I missing some option to inform make catkin the right path for this file ?

Ellon, 2019-07-03:
OK I found the problems.

One problem came from my in-house modified installation of ROS that puts the setup files on `${INSTALL_SPACE}/etc/ros` instead of `${INSTALL_SPACE}/env.sh` (I thought this could be selected by ROS somehow). This change breaks the installation of the packages, so I cannot use `--install` in the compile command line and I have to work in the devel space only.

The other problem came again from this in-house installation that is a little outdated and miss some fixes, notably the one that allow plugins to be declared using catkin package XML format 2 files (see [this question](http://answers.ros.org/question/191889/cannot-export-nodelet-plugins-in-packagexml-format-2/) for a similar problem using nodelets). Changing the XML file to format 1 solved the problem.

closed 2020-10-17 by MichaelGrupp