cartographer-project/cartographer_ros #1288: How to calibrate the extrinsic parameters between two Lidars ?

MaxChanger, 2019-05-15:
Hello, I want to use two Velodyne-16 Lidars in cartographer_ros now (one horizontal and one vertical ), but I can't know exactly the extrinsic parameters between two Lidars , that is, how to calibrate the relative positions of the two Lidars accurately.
Although I think this may be beyond the scope of this project, I cannot find a similar Lidars calibration problem on the network.
I would appreciate it very much if you could give me some advice.
Thanks!

gaschler, 2019-05-16:
Just a small note, correct, Cartographer has no features to correct/calibrate lidar extrinsics. (It only has a feature to optimize the IMU rotation.)

MaxChanger, 2019-05-16:
Ok, thank you for your reply.
But I remember that the cartographer supports multiple lidar access, so are they dependent on the availability of extrinsic parameters?

Ly0n, 2019-05-17:
You could try this tool for finding the extrinsic parameters: 
https://github.com/ethz-asl/lidar_align

> Note: Accurate results require highly non-planar motions, this makes the technique poorly suited for calibrating sensors mounted to cars.

MaxChanger, 2019-05-17:
> You could try this tool for finding the extrinsic parameters:
> https://github.com/ethz-asl/lidar_align
> 
> > Note: Accurate results require highly non-planar motions, this makes the technique poorly suited for calibrating sensors mounted to cars.

Unfortunately, I can't quite understand how ethz's project applies to the relative position calibration of two Lidars.
Do you mean to use IMU as a mediator to calibrate two Lidars?

Ly0n, 2019-05-17:
> Do you mean to use IMU as a mediator to calibrate two Lidars?
Yes that would be the ideal way for the cartographer from my point of view.

Another way to get the transformation between both scans is to align both scan via ICP or something similar. If you don't want to program this on your own you could use the libpointmatcher for that:
https://github.com/ethz-asl/libpointmatcher/blob/master/doc/Introduction.md

MaxChanger, 2019-05-17:
OK, Thanks for your reply! @Ly0n 
I will consider how to implement it on your recommendation.

closed 2020-10-27 by MichaelGrupp