cartographer-project/cartographer_ros #1293: I can't convert pbstream file into yaml file. Can I help you solve it?

ytgcljj, 2019-05-20:
i run this 
nuc16@nuc16:~$ roslaunch cartographer_ros offline_pbstream_to_rosmap.launch pbstream_filename:=/home/nuc16/test.pbstream map_filestem:=/home/nuc16/ros_map resolution:=0.05

... logging to /home/nuc16/.ros/log/2efe7a32-7ac0-11e9-a70a-00c2c6bb6575/roslaunch-nuc16-11707.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nuc16:39931/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    cartographer_pbstream_to_ros_map (cartographer_ros/cartographer_pbstream_to_ros_map)

ROS_MASTER_URI=http://localhost:11311

process[cartographer_pbstream_to_ros_map-1]: started with pid [11732]
I0520 17:50:53.276638 11732 pbstream_to_ros_map_main.cc:52] Loading submap slices from serialized data.
[cartographer_pbstream_to_ros_map-1] process has died [pid 11732, exit code -11, cmd /home/nuc16/catkin_ws/install_isolated/lib/cartographer_ros/cartographer_pbstream_to_ros_map -pbstream_filename /home/nuc16/test.pbstream -map_filestem /home/nuc16/ros_map -resolution 0.05 __name:=cartographer_pbstream_to_ros_map __log:=/home/nuc16/.ros/log/2efe7a32-7ac0-11e9-a70a-00c2c6bb6575/cartographer_pbstream_to_ros_map-1.log].
log file: /home/nuc16/.ros/log/2efe7a32-7ac0-11e9-a70a-00c2c6bb6575/cartographer_pbstream_to_ros_map-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

ytgcljj, 2019-05-20:
[roslaunch][INFO] 2019-05-20 17:50:52,264: Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
[roslaunch][INFO] 2019-05-20 17:50:52,297: Done checking log file disk usage. Usage is <1GB.
[roslaunch][INFO] 2019-05-20 17:50:52,297: roslaunch starting with args ['/opt/ros/kinetic/bin/roslaunch', 'cartographer_ros', 'offline_pbstream_to_rosmap.launch', 'pbstream_filename:=/home/nuc16/test.pbstream', 'map_filestem:=/home/nuc16/ros_map', 'resolution:=0.05']
[roslaunch][INFO] 2019-05-20 17:50:52,297: roslaunch env is {'TURTLEBOT_TYPE': 'turtlebot', 'ROS_DISTRO': 'kinetic', 'LESSOPEN': '| /usr/bin/lesspipe %s', 'SSH_CLIENT': '192.168.6.4 39041 22', 'LOGNAME': 'nuc16', 'USER': 'nuc16', 'HOME': '/home/nuc16', 'PATH': '/home/nuc16/catkin_ws/install_isolated/bin:/opt/ros/kinetic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin', 'CMAKE_PREFIX_PATH': '/home/nuc16/golo_ws/devel:/home/nuc16/catkin_ws/install_isolated:/opt/ros/kinetic', 'LD_LIBRARY_PATH': '/home/nuc16/golo_ws/devel/lib:/home/nuc16/catkin_ws/install_isolated/lib:/opt/ros/kinetic/lib:/opt/ros/kinetic/lib/x86_64-linux-gnu', 'XDG_RUNTIME_DIR': '/run/user/1000', 'LANG': 'zh_CN.UTF-8', 'TERM': 'xterm', 'SHELL': '/bin/bash', 'LANGUAGE': 'zh_CN:zh', 'TURTLEBOT_STACKS': 'hexagons', 'SHLVL': '1', 'QT_QPA_PLATFORMTHEME': 'appmenu-qt5', 'TURTLEBOT_SIMULATION': 'false', 'ROS_ETC_DIR': '/opt/ros/kinetic/etc/ros', 'ROS_LOG_FILENAME': '/home/nuc16/.ros/log/2efe7a32-7ac0-11e9-a70a-00c2c6bb6575/roslaunch-nuc16-11707.log', 'TURTLEBOT_RAPP_PACKAGE_WHITELIST': '[rocon_apps, turtlebot_rapps]', 'TURTLEBOT_BATTERY': '/sys/class/power_supply/BAT0', 'TURTLEBOT_INTERACTIONS_LIST': '[turtlebot_bringup/admin.interactions, turtlebot_bringup/documentation.interactions, turtlebot_bringup/pairing.interactions, turtlebot_bringup/visualisation.interactions]', 'TURTLEBOT_RAPP_PACKAGE_BLACKLIST': '[]', 'PYTHONPATH': '/home/nuc16/golo_ws/devel/lib/python2.7/dist-packages:/home/nuc16/catkin_ws/install_isolated/lib/python2.7/dist-packages:/opt/ros/kinetic/lib/python2.7/dist-packages:/opt/movidius/caffe/python', 'TURTLEBOT_SERIAL_PORT': '/dev/kobuki', 'TURTLEBOT_NAME': 'turtlebot', 'PKG_CONFIG_PATH': '/home/nuc16/golo_ws/devel/lib/pkgconfig:/home/nuc16/catkin_ws/install_isolated/lib/pkgconfig:/opt/ros/kinetic/lib/pkgconfig:/opt/ros/kinetic/lib/x86_64-linux-gnu/pkgconfig:/usr/local/lib/pkgconfig', 'ROS_PACKAGE_PATH': '/home/nuc16/golo_ws/src:/home/nuc16/catkin_ws/install_isolated/share:/opt/ros/kinetic/share', 'ROS_ROOT': '/opt/ros/kinetic/share/ros', 'XDG_SESSION_ID': '37', 'TURTLEBOT_BASE': 'kobuki', '_': '/opt/ros/kinetic/bin/roslaunch', 'SSH_CONNECTION': '192.168.6.4 39041 192.168.8.16 22', 'LESSCLOSE': '/usr/bin/lesspipe %s %s', 'SSH_TTY': '/dev/pts/6', 'ROSLISP_PACKAGE_DIRECTORIES': '/home/nuc16/golo_ws/devel/share/common-lisp', 'XDG_DATA_DIRS': '/usr/local/share:/usr/share:/var/lib/snapd/desktop', 'PWD': '/home/nuc16', 'ROS_MASTER_URI': 'http://localhost:11311', 'TURTLEBOT_3D_SENSOR': 'kinect', 'ROS_VERSION': '1', 'MAIL': '/var/mail/nuc16', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:'}
[roslaunch][INFO] 2019-05-20 17:50:52,297: starting in server mode
[roslaunch.parent][INFO] 2019-05-20 17:50:52,298: starting roslaunch parent run
[roslaunch][INFO] 2019-05-20 17:50:52,298: loading roscore config file /opt/ros/kinetic/etc/ros/roscore.xml
[roslaunch][INFO] 2019-05-20 17:50:52,928: Added core node of type [rosout/rosout] in namespace [/]
[roslaunch.config][INFO] 2019-05-20 17:50:52,928: loading config file /home/nuc16/catkin_ws/install_isolated/share/cartographer_ros/launch/offline_pbstream_to_rosmap.launch
[roslaunch][INFO] 2019-05-20 17:50:52,929: Added node of type [cartographer_ros/cartographer_pbstream_to_ros_map] in namespace [/]
[roslaunch][INFO] 2019-05-20 17:50:52,930: ... selected machine [] for node of type [cartographer_ros/cartographer_pbstream_to_ros_map]
[roslaunch.pmon][INFO] 2019-05-20 17:50:52,931: start_process_monitor: creating ProcessMonitor
[... cut ...]
[roslaunch][INFO] 2019-05-20 17:50:53,671: shutting down processing monitor <ProcessMonitor(ProcessMonitor-1, stopped daemon 139897168688896)>
[roslaunch.pmon][INFO] 2019-05-20 17:50:53,672: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 139897168688896)>
[roslaunch][INFO] 2019-05-20 17:50:53,672: ... shutting down processing monitor complete
[roslaunch][INFO] 2019-05-20 17:50:53,672: done
[roslaunch.pmon][INFO] 2019-05-20 17:50:53,672: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 139897168688896)>
[rospy.core][INFO] 2019-05-20 17:50:53,673: signal_shutdown [atexit]

ytgcljj, 2019-05-20:
this is launch file
<launch>
  <node name="cartographer_pbstream_to_ros_map" pkg="cartographer_ros"
      type="cartographer_pbstream_to_ros_map" args="
          -pbstream_filename $(arg pbstream_filename)
          -map_filestem $(arg map_filestem)
          -resolution $(arg resolution)"
      output="screen">
  </node>
</launch>

ytgcljj, 2019-05-20:
<launch>
  <node name="cartographer_pbstream_to_ros_map" pkg="cartographer_ros"
      type="cartographer_pbstream_to_ros_map" args="
          -pbstream_filename $(arg pbstream_filename)
          -map_filestem $(arg map_filestem)
          -resolution $(arg resolution)"
      output="screen">
  </node>
</launch>

artemiialessandrini, 2019-09-27:
rosrun cartographer_ros cartographer_pbstream_to_ros_map -map_filestem=/home/hostname/mymap -pbstream_filename=/home/hostname/Downloads/mymap.pbstream -resolution=0.05

closed 2020-10-27 by MichaelGrupp