cartographer-project/cartographer_ros #1294: Manually registering PointClouds using optimized RT(rotation-translation) from 2d lidar cartographer

UditSinghParihar, 2019-05-21:
Hello sir,
    I am using 2d lidar cartographer and getting a good 2d map.   
Now I also have PointCloud topics along with the lidar topic in my bagfile, so I would like to register the PointClouds based on the RT which I got from pbstream file at the end of the run.  
I would be require to post process on the obtained PointCloud based on the my deep learning framework, so I need independent access to PointClouds and the corresponding RTs between them.

Could you guide me to the part of the cartographer_ros codebase where it sample the lidar scan used as a node for graph slam, so that I could extract the corresponding PointCloud at that particular timestamp?
Please let me know if it is possible?

ojura, 2019-05-21:
You have sparsified versions of the observed scan inside a node's `ConstantData`. If you want the full cloud, you will unfortunately have to process the source dataset bag again using the poses from the serialized mapping result (in a .pbstream). Cartographer's `assets_writer` does exactly this to perform mapping with known poses.

Or, you could hack Cartographer to store the full point cloud inside the node constant `Data` [here](https://github.com/googlecartographer/cartographer/blob/bcd5486025df4f601c3977c44a5e00e9c80b4975/cartographer/mapping/internal/2d/local_trajectory_builder_2d.cc#L294). This was actually stored it at one point, but it was removed in order to reduce memory consumption (I was trying to oppose this, but was not successful :-) ).

UditSinghParihar, 2019-05-21:
Hi,
    I would use your first approach, currently I am using the tool [pbstream_trajectories_to_rosbag_main](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/cartographer_ros/dev/pbstream_trajectories_to_rosbag_main.cc), to get the rosbag file of the optimized tf topic stored in pbstream.    
I would run this new bagfile along with my old bagfile containing PointClouds topics and get the synchronized topics and register it accordingly. I would update if any any issue arrives.

Thanks

athtest800, 2019-06-03:
Do you have success publishing other pointclouds with the corrected trajectory, that are in the same bag with other topic name?

UditSinghParihar, 2019-06-03:
Hi,.   
I haven't been out for sometime, so haven't register pointclouds. But I got the bagfile containing the optimized trajectory from cartographer having the same timestamp as the original bagfile. So this could be registered manually using synchronisation (filter) ros library and PCL.

athtest800, 2019-06-03:
Thank you for the answer. Is it possible to describe more on how to use this "pbstream_trajectories_to_rosbag_main" ? Do you have a sample file (.launch) on how to use it?

UditSinghParihar, 2019-06-03:
First check whether you have the binary for `pbstream_trajectories_to_rosbag_main` in your machine, if not then build cartographer_ros again from the master branch.   
After building you can directly use the binary.    
To search for any file :-.     
`sudo find / -iname "pbstream_trajectories_to_rosbag_main"`

UditSinghParihar, 2019-06-07:
Hi @ojura ,  
1. Could you tell whether the output bagfile from [pbstream_trajectories_to_rosbag_main](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/cartographer_ros/dev/pbstream_trajectories_to_rosbag_main.cc), contains the last optimized trajectory when the cartographer finishes or it contains the trajectory when the cartographer was running(i.e unoptimized)?    
2. I tried to extract the trajectory and register the PointClouds from output bagfile(from `pbstream_trajectories_to_rosbag_main`), here is the result :     
[image]
[image]
[Code](https://github.com/UditSinghParihar/ROS_WS/blob/master/pcd_extract/src/carto_cloud.cpp) for registering pointclouds from trajectory from output bagfile.

ojura, 2019-06-09:
Both, it is optimized while Cartographer is running.

Also, have a look at the tuning guide - you have local SLAM making rough errors and should try tuning that first. It is futile to try optimizing if you have such bad local quality - optimization is meant to eliminate slowly accumulating drift in large loops, not to correct rough local errors.

UditSinghParihar, 2019-06-09:
Ok, first I would try to get good local maps, then try to get optimized trajectory. I think my robot have very bad odometry, so removing it may improve the maps. And then would comment on the optimized trajectory.       
Thanks

UditSinghParihar, 2019-06-11:
Hi @ojura ,
    Could you tell me some parameters to tune local slam . I am only using Lidar sensor (as my wheel odometry is faulty). The trajectory from cartographer is sometimes coming to be in backward direction, but in reality robot is moving forward(may be repeated rack space).  Here is a [video](https://drive.google.com/open?id=1fDDqP3Uj9paVN8zIpgINfkD0J6w5g0Jd) showing the problem.   
And [here](https://gist.github.com/UditSinghParihar/ef03cc911695ce65d6ad716d458bb0d4) are my configuration files.    
Is it possible to tune the parameters for good local slam or problem is because of the repeated structures in my environment?

ojura, 2019-06-12:
I can see that your lidar can have problems with movement observability because it sees a long monotonic hallway with no features... I'd say that you need a wider-fov, longer range lidar. Or an environment with more features. Unfortunately, tuning local slam is more or less lots of trial and error, and I cannot give you specific advice. Try enabling the online correlative scan matcher. The 2D TSDF scan matcher might also be of help, I have not used it though.

closed 2020-10-24 by UditSinghParihar