cartographer-project/cartographer_ros #1296: NavSat Implementation

Karltlbm, 2019-05-21:
Hi !

I am now adding GNSS positionning to my cartographer_ros implementation.

I use navsat fix messages that I acquired from an antenna and reciever.

In cartographer 3D that used to work without it , I changed these lines:

launch file:
`<remap from="fix" to="navsat_pos" />`

`<node pkg="tf" type="static_transform_publisher" name="tf2" args="0 0 0 0 0 0  base_link navsat_link 50" />`

lua file:
`use_nav_sat = true`

The result is this:

```
[ WARN] [1558448912.285468952, 1558447118.718096664]: W0521 16:28:32.000000 15402 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument target_frame does not exist. 
[ INFO] [1558448912.285897776, 1558447118.719421622]: I0521 16:28:32.000000 15402 sensor_bridge.cc:88] Using NavSatFix. Setting ecef_to_local_frame with lat = 12.6139, long = 1.43339.
[ WARN] [1558448912.288218687, 1558447118.822027745]: W0521 16:28:32.000000 15402 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument target_frame does not exist. 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[ WARN] [1558448912.291843581, 1558447118.921378049]: W0521 16:28:32.000000 15402 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1558448912.294411677, 1558447119.017106302]: W0521 16:28:32.000000 15402 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1558448912.296625682, 1558447119.107034573]: W0521 16:28:32.000000 15402 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1558448912.298618946, 1558447119.206882088]: W0521 16:28:32.000000 15402 tf_bridge.cc:52] "imu_link" passed to lookupTransform argument target_frame does not exist.
```
And a lot more of this warning to finish by killing cartographer...
So navsat topic is seen but it doesn't work.

What did I do wrong ?

lucafei, 2022-01-23:
hi， can you share how to add gps in cartographers 2d mapping and pure localization ?

MikeChie, 2022-03-23:
Hi Karltlbm, do you have gps rosbag data shared?

Karltlbm, 2022-03-23:
Hi all, thanks for your help but I stopped working on this topic. Good luck :)