cartographer-project/cartographer_ros #1297: Can we know at which frame each trajectory node was generated?

nobinov, 2019-05-22:
Hi all!
Currently i am trying to extract the final trajectory of a robot after doing offline mapping in cartographer. By following issue #332 , and using this [code](https://gist.github.com/nobinov/594b1b54e60486adcb1c3c513c74c887), now I am able to access the pose of robot for each trajectory node.

I ran my cartographer_ros in my bagfile that generated from [KITTI raw dataset](http://www.cvlibs.net/datasets/kitti/raw_data.php) using [kitti2bag](https://github.com/tomas789/kitti2bag) and have 1101 frames on it. From the .pbstream file that generated, I got 526 trajectory nodes. From the number itself, it seems there's no correlation between amount of generated trajectory nodes and amount of scanning frames.

Is there any way to know at which frame each trajectory node was generated? Or what parameter(s) that can affect the amount of generated nodes?

Thank you!

MichaelGrupp, 2019-05-27:
> From the number itself, it seems there's no correlation between amount of generated trajectory nodes and amount of scanning frames.

To get the same number of poses as there are scans, you have to make sure that a trajectory node is inserted for every laser scan:

* disable [motion filter](https://google-cartographer.readthedocs.io/en/latest/configuration.html#cartographer-mapping-proto-motionfilteroptions) (set values to 0)
* make sure that `num_accumulated_range_data` equals `options.num_laser_scans * options.num_subdivisions_per_laser_scan`, i.e. using single scans for scan matching

Maybe there are other options you need to change that I forgot, but you can try it.

GuillermoHerreraF, 2019-06-02:
Hey @nobinov 
I'm working on a project where I need to achieve the same task you are describing (extract the poses from a pbstream file generated with cartographer), and I'm also working with the KITTI dataset. I have several questions regarding where should I place the pbstream_reader.cpp source code within my workspace. 
I believe this is the correct path, but I am not completely sure:
catkin_ws -> src -> cartographer_ros -> cartographer_ros -> cartographer_ros -> pbstream_reader.cpp

Also, the CMakeLists.txt file I need to modify is the one in the following path?
catkin_ws -> src -> cartographer_ros -> cartographer_ros ->  CMakeLists.txt

The modifications that need to be done for the above mentioned CMakeLists.txt file are adding the following lines? 
add_executable(read pbstream_reader.cpp)
target_link_libraries(read cartographer_ros )

And finally, is this file run as a ros node (rosrun)? I am still new to ROS so sorry if I'm asking something obvious. 

Thank you in advance!

MichaelGrupp, 2019-06-03:
@GuillermoHerreraF There's already something similar in the repository: https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/cartographer_ros/dev/pbstream_trajectories_to_rosbag_main.cc

```
rosrun cartographer_ros cartographer_dev_pbstream_trajectories_to_rosbag ...
```

GuillermoHerreraF, 2019-06-03:
Hello @MichaelGrupp thank you for your help. 

I tried adding the pbstream_trajectories_to_rosbag_main.cc to my cartographer workspace and modified the corresponding CMakeLists.txt following this:
https://code.research.uts.edu.au/13110756/cartographer_ros/commit/6ee68c22f922c6234c5488f649f1b413e860eae2

But it gives me a compilation error. I also tried cloning the git from the link you provided into my workspace, but I still get compilation errors. 

How should I modify my workspace in order to be able to run the executable with the command you provided?

Thank you again.

GuillermoHerreraF, 2019-06-04:
@MichaelGrupp I manage to install the new version of cartographer and cartographer_ros in a new workspace. but I had to do it manually (without the cartographer_ros.rosinstall) and everything compiles.

Now I'm trying to run the command rosrun cartographer_ros cartographer_dev_pbstream_trajectories_to_rosbag [pbstream file] [output bag file], but it gives me an error saying that the input pbstream file is missing. What is the format to pass the pbstream and bag files?

MichaelGrupp, 2019-06-04:
`cartographer_ros cartographer_dev_pbstream_trajectories_to_rosbag --input <pbstream> --output <bagfile>`

GuillermoHerreraF, 2019-06-04:
Great thank you very much for your help @MichaelGrupp!

nobinov, 2019-06-07:
hi @MichaelGrupp and @GuillermoHerreraF , thank you for replying this issue. glad to get a new insight for this problem!

>  disable motion filter (set values to 0)
    make sure that num_accumulated_range_data equals options.num_laser_scans * options.num_subdivisions_per_laser_scan (or num_multi_echo_laser_scans, depending on what's used...) i.e. using single scans for scan matching

I've tried this, and now the generated trajectories increased, but only to 1090. Meanwhile I have 1102 frames. I am still not check whether it can be matched to the frames in the rosbag or not, I'll try it and give you an update soon.

> I tried adding the pbstream_trajectories_to_rosbag_main.cc to my cartographer workspace and modified the corresponding CMakeLists.txt following this:
https://code.research.uts.edu.au/13110756/cartographer_ros/commit/6ee68c22f922c6234c5488f649f1b413e860eae2
But it gives me a compilation error. I also tried cloning the git from the link you provided into my workspace, but I still get compilation errors.

In my environment, the error seems ocurred because I am using the old version of cartographer_ros and there's no package named Abseil that required when calling `#include "absl/strings/str_cat.h"` I've tried to fix it by install those package manually using `sudo apt-get install ros-${ROS_DISTRO}-abseil-cpp ` and with another way by include the Abseil as ros package based on [https://github.com/Eurecat/abseil-cpp](https://github.com/Eurecat/abseil-cpp). After that, I modified CMakeLists.txt on `src/cartographer_ros/cartographer_ros` by adding `find_package(abseil_cpp REQUIRED)`. However, both of them are still not detected. 

Is there any way to fix those problem without re-installing cartographer_ros from scratch again? Because I've made several modification needed internally on cartographer-ros.

Thank you!

GuillermoHerreraF, 2019-06-11:
Hey @nobinov. The only way i could make it work was by installing from scratch the cartographer and cartographer_ros packages. As you pointed out, if not, it does not find the Abseil library. I tried to modify the files without having to reinstall both packages but I couldn't make it work as I mentioned earlier. 

I wanted to ask you how did you manage to increase the number of poses to scans? I tried doing what @MichaelGrupp suggested, modifying the max_time_seconds, max_distance_meters and max_angle_radians variables to 0 in the motion_filter_options.proto file; however, I get this compilation error when I recompile it:

[image]

Also, which value did you put in the num_accumulated_range_data variable from the local_trajectory_builder_options_3d.proto file? I changed to 0 as well as the num_laser_scans from my lua file is 0. I do not know if I understood correctly.

MichaelGrupp, 2019-06-12:
Don't modify the .proto files, they define internal message formats. If you want to change the configuration, adapt the .lua files.

GuillermoHerreraF, 2019-06-12:
@MichaelGrupp I did as you suggested and indeed I get more poses than before. However, I checked poses from the same timestamp (with and without the .lua modifications) and they do not agree. 

This is the changes I made in the .lua:

TRAJECTORY_BUILDER_3D.motion_filter.max_time_seconds = 0
TRAJECTORY_BUILDER_3D.motion_filter.max_distance_meters = 0
TRAJECTORY_BUILDER_3D.motion_filter.max_angle_radians = 0

TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 0

as 

num_laser_scans = 0,
num_subdivisions_per_laser_scan = 1,

Do you have any idea why this difference appears?

nobinov, 2019-06-13:
@MichaelGrupp @GuillermoHerreraF  I did tried to install cartographer from master branch, but I got an error when compiling them : [https://github.com/googlecartographer/cartographer/issues/1589](https://github.com/googlecartographer/cartographer/issues/1589) do you guys have any idea what's missing?

GuillermoHerreraF, 2019-06-13:
Hey @nobinov. Did you install it using the cartographer_ros.rosinstall file? Because I couldn't make it work that way... for some reason it keeps installing the older version. What I did was copy the folders corresponding to cartographer and cartographer_ros packages within my workspace (https://github.com/googlecartographer/cartographer_ros and https://github.com/googlecartographer/cartographer) and recompile everything with catkin_make_isolated --install --use-ninja.

The problem with this for you is that you will have to modify again the changes you made before in those packages. 

I wanted to ask you, can you shared the .lua configuration file that you are using for the KITTI dataset please. 

Let me know if it works for you.

nobinov, 2019-06-13:
@GuillermoHerreraF you can see my simple configuration on here [https://github.com/nobinov/kitti_cartographer](https://github.com/nobinov/kitti_cartographer). it's working but not fully tuned so the result might be a bit off.

I also already tried to install the way you already did. But in my case, the `cartographer` package got an error out of nowhere on ray_casting.cc, so it failed to compile. (error messages can be seen on [https://github.com/googlecartographer/cartographer/issues/1589](https://github.com/googlecartographer/cartographer/issues/1589) ). Did you previously got the similar error while compiling?

nobinov, 2019-09-09:
hi @MichaelGrupp previously i am working on the kitti data set that only have one pointcloud2 topic. Now I am using another data that use 3 different pointcloud2 topics, and 1 of them is using different frequency(2 slow velodynes, 1 fast velodyne). Is there any way to set the frame references to one specific frequency?

MichaelGrupp, 2020-10-16:
I'm sorry, I can't help with setting this up.

closed 2020-10-16 by MichaelGrupp

marcoluis97, 2022-12-02:
Greetings @nobinov @MichaelGrupp,

> I've tried this, and now the generated trajectories increased, but only to 1090. Meanwhile I have 1102 frames. I am still not check whether it can be matched to the frames in the rosbag or not, I'll try it and give you an update soon.

were you ever able to understand why this mismatch between the number of /scan topics and the number of /trajectory topics (obtained from running `cartographer_dev_pbstream_trajectories_to_rosbag`)?