cartographer-project/cartographer_ros #1302: Start Cartographer Node without starting a trajectory

dwes7, 2019-05-28:
Setup:
- Small Quadrotor
- onboard Intel NUC
- Lidar
- VIO from realsense T265
- IMU
- Pixracer
- Arducopter
- Cartographer running 2D SLAM ( this is a requirement of the project)

What am I trying to do? 
When set to AutoMode, quadrotor will launch, fly an autonomous exploratory mission and return home when complete.

Without going too much into the details of my project, I'd simply like to know if there is any way to start the cartographer node without starting a trajectory. The issue I'm having is that so far, all my ROS nodes are ready and running (including cartographer node) before the quad is flipped into AutoMode and takes off. When the cartographer node starts with the quad on the ground, a trajectory is started and a submap is built. Since the Laser scan of the environment is different from the ground than in the air, we then end the trajectory, take off to desired altitude, and start a new trajectory. When running 2D SLAM, that first trajectory that was started on the ground is taken into account and messes up the building of the map, and thus our localization in it.

Some options of what I would like to do is start cartographer without starting an initial trajectory, flip the quad into automode, then programmatically start a trajectory once altitude is reached. Or I would like to know if there's a way for cartographer to ignore or delete one of the trajectories when building a map. Are either of these things possible?

Please consider my questions. Thank you all for the great work in building and maintaining this tool!

ojura, 2019-05-28:
You can pass the flag `-start_trajectory_with_default_topics=false`.

Also try running `-help` to see a list of other flags.

dwes7, 2019-05-28:
Thanks for the quick response! I'll give this a try and report back.

dwes7, 2019-05-28:
That worked! Just had to modify the arguments when running the cartographer_start_trajectory node to not include the initial position because it was referencing a trajectory that didn't exist.

Thanks for you help!

closed 2019-05-28 by dwes7

artemiialessandrini, 2019-11-18:
> That worked! Just had to modify the arguments when running the cartographer_start_trajectory node to not include the initial position because it was referencing a trajectory that didn't exist.
> 
> Thanks for you help!

That's actually quite an interesting thing.
Can we run `/start_trajectory `with` -start_trajectory_with_default_topics` to be `true`?
Then it's possible to set initial_pose: {x:   ,y:   ,z:   ,} {r:   ,p:   ,y:   ,w:   }

It might be the way to start trajectory not in [0, 0] pose on the map frame, or am I wrong?
Maybe somehow set `relative_to_trajectory_id:` to `origin`