cartographer-project/cartographer_ros #1303: range_data_collator.cc Dropped x earlier points.

narogon, 2019-05-28:
Hi everyone, 

Could anybody help me explaining why I receive this warning? Is this important? 
[ WARN] [1559070323.023164914]: W0528 21:05:23.000000 10756 range_data_collator.cc:76] Dropped 612 earlier points.

This is the rosbag_validate output:
https://gist.github.com/narogon/c44f79c92ea76a1efead15d82182d809

And here are the launch, lua, bag and urdf configuration files
https://www.dropbox.com/sh/kr6ej4hlbemhd3w/AADCpu2tDLm5mmiQZF5lZXzya?dl=0

narogon, 2019-06-13:
Isn't there anyone here who can help me?

ojura, 2019-06-13:
This could indicate a problem with your timestamps. When a scan is processed, if some of its points end up having an earlier timestamp than the end of the previous scan (i.e. they overlap in time with the previous scan), Cartographer will discard these points.

narogon, 2019-06-17:
The times of my laser scans messages are always increasing. I've checked it analyzing the times of the topic echos. 
Also my laser scan messages are always 2400 points length but the warning messages says that dropped a variable quantity of them...

Is there any other test that I could do to check this?

ojura, 2019-06-17:
It is not enough to just compare the single stamps. You have to consider the duration of the scan, i.e. `scan_time` and/or 2400 * `time_increment`. Your rosbag validate output shows that the scans are clearly overlapping in time.

narogon, 2019-06-20:
My laser msg header stamp is created using: 
`scanmsg.header.stamp = ros::Time::now();`
I've also tried to use the timestamp that lidar sends and also the lidar time plus a difference with ros time (catched the first time) with even worse results.

How could I solve it? Or  must I only forget the warnings?

ojura, 2019-06-23:
Hardware trigger timesync would be most preferable. If not, some kind of timestamp filtering which will sync the sensor clock with the system clock(`ros::Time::now`), while avoiding sensor clock drift and the system clock jitter due to OS scheduling.

closed 2020-10-16 by MichaelGrupp

csufangyu, 2020-12-09:
hi I get a same error and I use the R2000,could you tell me how to solve it?

iLanceS, 2022-03-04:
> hi I get a same error and I use the R2000,could you tell me how to solve it?

me too

Gunz-katomaran, 2022-11-07:
Hi guys, I have the same issue.. is anyone help me out