cartographer-project/cartographer_ros #1304: Topics misunderstanding problem

hachbani2019, 2019-05-29:
Hi,
I'm tying to setup cartographer_ros, I quote from the documentation : 

> If you use only one sensor_msgs/PointCloud2 topic, it is expected be named points2. If you have more, they should be named points2_1, points2_2, etc…

I used a Lidar (Ouster) to record some data to a bag file, my bag file publish data to /os1_cloud_node/points. According to the quote, I did add this line to my launch file: 
`<remap from="/os1_cloud_node/points" to="points2"/>
`
First question, is what I did right ? Because It doesn't seem to work, when I run the roslaunch command, It doesn't display anything, and it doesn't give me an error neither. SO what might be the problem ? 

Thanks

MrGitGo, 2019-06-03:
If you do the remap in the roslaunch for the cartographer, you have to write this: <remap from="points2" to="/os1_cloud_node/points"/>

dsb6063, 2019-07-03:
Did that remap solve your issue

hachbani2019, 2019-07-05:
Hi, the remap was just a part of the problem. It's not what made cartographer run on my Ouster BAG file. But, I managed after trying different fixes to get it to work.

closed 2019-07-05 by hachbani2019