cartographer-project/cartographer_ros #1305: rosdep install step fails.

alankdkd, 2019-05-29:
1. Full output:  rosbag_validate: command not found
2 & 3. I didn't get far enough to launch or use any bag files

I've followed all the steps to install cartographer (repeatedly), but it fails on the step:

rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y

I get the following output:

dpkg-query: no packages found matching python-sphinx
dpkg-query: no packages found matching libqt5widgets5
dpkg-query: no packages found matching liblua5.2-dev
dpkg-query: no packages found matching libboost-all-dev
dpkg-query: no packages found matching libpcl-dev
dpkg-query: no packages found matching libprotoc-dev
dpkg-query: no packages found matching gfortran
dpkg-query: no packages found matching qtbase5-dev
dpkg-query: no packages found matching libgflags-dev
dpkg-query: no packages found matching libeigen3-dev
dpkg-query: no packages found matching libprotobuf-dev
dpkg-query: no packages found matching libcairo2-dev
dpkg-query: no packages found matching libqt5core5a
dpkg-query: no packages found matching libsuitesparse-dev
dpkg-query: no packages found matching protobuf-compiler
dpkg-query: no packages found matching libgoogle-glog-dev
dpkg-query: no packages found matching libatlas-base-dev
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies (ROS distro is not set. Make sure `ROS_DISTRO` environment variable is set, or use `--rosdistro` option to specify the distro, e.g. `--rosdistro indigo`):
cartographer_ros_msgs: Cannot locate rosdep definition for [std_msgs]
cartographer: Cannot locate rosdep definition for [catkin]
cartographer_ros: Cannot locate rosdep definition for [rosunit]
cartographer_rviz: Cannot locate rosdep definition for [rviz]
ceres-solver: Cannot locate rosdep definition for [catkin]

I'm using Ubuntu 18.04.  What did I do wrong?

Thanks!
Alan

MagicianCH, 2019-05-30:
Hi,have you installed ROS yet, for example ROS melodic?
if yes, please make sure run "source  /opt/ros/melodic/setup.bash" first.

alankdkd, 2019-05-30:
Thanks MagicianCH, I'll try it.

alankdkd, 2019-05-30:
SOLVED.  The problem was ${ROS_DISTRO} needed to be replaced by the distro name, e.g. melodic.

closed 2019-05-30 by alankdkd