cartographer-project/cartographer_ros #1306: Missing .launch files and packages

alankdkd, 2019-05-30:
1. rosrun cartographer_ros cartographer_rosbag_validate -bag_filename cartographer_paper_deutsches_museum.bag

Full output: [rospack] Error: package 'cartographer_ros' not found

2 & 3. I haven't reached the point where I can process a bag file; not getting that far.
============================

I've followed the steps up to https://google-cartographer-ros.readthedocs.io/en/latest/demos.html, but rosrun doesn't find any packages, and there are no .launch files on my system.

It appears the demo installation was incomplete.  How does one obtain/install packages and .launch files?

Thanks!
Alan

MrGitGo, 2019-06-03:
did you source your packages?

alankdkd, 2019-06-04:
Hi MrGitGo, I'm not sure what you mean by "source your packages".  I'm from a Windows background, and Linux is new to me.

alankdkd, 2019-06-21:
Folder problem.  Fixed.  Closing issue.

closed 2019-06-21 by alankdkd