cartographer-project/cartographer_ros #1310: Non-catkin packages after installing cartographer?

apletta, 2019-06-05:
Having trouble sourcing my workspace. It seems I somehow made a workspace out of my src folder within my catkin_ws workspace. I'm thinking this originated from when I ran "wstool init src". I followed all of the steps in the [Cartographer ROS for TurtleBots](https://google-cartographer-ros-for-turtlebots.readthedocs.io/en/latest/)

When I run
`source install_isolated/setup.bash`
I can launch the cartographer packages I have but not the other packages in my workspace.

When I run
`source /opt/ros/melodic/setup.bash`
and
`source ~/catkin_ws/devel/setup.bash`
I can launch everything in my workspace but not the cartographer packages. 

I have all of these source commands in my ~/.bashrc but noticed it appears that only the last/farthest down source command is applied.

I can run the demos from that Cartographer ROS for TurtleBots page (linked above) but can't source the cartographer files and my own workspace files at the same time. 

My workspace is setup as follows. Installing Cartographer through the tutorial (linked above) added build_isolated, devel_isolated, and install_isolated into the first level and catographer, cartographer_ros, cartographer_turtlebot, and ceres-solver into the second level. I'll include a picture to help describe.

[image]

Running catkin_make tells me that there is a non-catkin package in the workspace:

===================== output code vvv
`alex@alex-UbuntuROS:~/catkin_ws$ catkin_make
Base path: /home/alex/catkin_ws
Source space: /home/alex/catkin_ws/src
Build space: /home/alex/catkin_ws/build
Devel space: /home/alex/catkin_ws/devel
Install space: /home/alex/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/alex/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/alex/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/alex/catkin_ws/devel;/opt/ros/melodic
-- This workspace overlays: /home/alex/catkin_ws/devel;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.15", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/alex/catkin_ws/build/test_results
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.15") 
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
WARNING: package "cartographer_turtlebot" should not depend on metapackage "kobuki_core" but on its packages instead
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 25 packages in topological order:
-- ~~  - ceres-solver (plain cmake)
-- ~~  - cartographer (plain cmake)
-- ~~  - my_simulations
-- ~~  - openslam_gmapping
-- ~~  - slam_gmapping (metapackage)
-- ~~  - cartographer_ros_msgs
-- ~~  - testrobot_navigation
-- ~~  - testrobot_slam
-- ~~  - testrobot_teleop
-- ~~  - turtlebot3 (metapackage)
-- ~~  - turtlebot3_msgs
-- ~~  - turtlebot3_navigation
-- ~~  - beginner_tutorials
-- ~~  - gazebo_testing
-- ~~  - testrobot_gazebo
-- ~~  - gmapping
-- ~~  - sick_scan
-- ~~  - turtlebot3_bringup
-- ~~  - turtlebot3_example
-- ~~  - turtlebot3_slam
-- ~~  - turtlebot3_teleop
-- ~~  - cartographer_ros
-- ~~  - cartographer_rviz
-- ~~  - cartographer_turtlebot
-- ~~  - turtlebot3_description
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:63 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/alex/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/alex/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:2966: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed`
===================== output code ^^^

When I run catkin_make_isolated I see the following:

===================== output code vvv
`alex@alex-UbuntuROS:~/catkin_ws$ catkin_make_isolated
Base path: /home/alex/catkin_ws
Source space: /home/alex/catkin_ws/src
Build space: /home/alex/catkin_ws/build_isolated
Devel space: /home/alex/catkin_ws/devel_isolated
Install space: /home/alex/catkin_ws/install_isolated
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 25 packages in topological order:
~~  - beginner_tutorials
~~  - cartographer_ros_msgs
~~  - ceres-solver (plain cmake)
~~  - cartographer (plain cmake)
~~  - cartographer_ros
~~  - cartographer_rviz
~~  - cartographer_turtlebot
~~  - gazebo_testing
~~  - my_simulations
~~  - openslam_gmapping
~~  - gmapping
~~  - sick_scan
~~  - slam_gmapping
~~  - testrobot_gazebo
~~  - testrobot_navigation
~~  - testrobot_slam
~~  - testrobot_teleop
~~  - turtlebot3
~~  - turtlebot3_description
~~  - turtlebot3_msgs
~~  - turtlebot3_bringup
~~  - turtlebot3_example
~~  - turtlebot3_navigation
~~  - turtlebot3_slam
~~  - turtlebot3_teleop
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

==> Processing catkin package: 'beginner_tutorials'
==> cmake /home/alex/catkin_ws/src/beginner_tutorials -DCATKIN_DEVEL_PREFIX=/home/alex/catkin_ws/devel_isolated/beginner_tutorials -DCMAKE_INSTALL_PREFIX=/home/alex/catkin_ws/install_isolated -G Unix Makefiles in '/home/alex/catkin_ws/build_isolated/beginner_tutorials'
CMake Error: Error: generator : Unix Makefiles
Does not match the generator used previously: Ninja
Either remove the CMakeCache.txt file and CMakeFiles directory or choose a different binary directory.
<== Failed to process package 'beginner_tutorials': 
  Command '['cmake', '/home/alex/catkin_ws/src/beginner_tutorials', '-DCATKIN_DEVEL_PREFIX=/home/alex/catkin_ws/devel_isolated/beginner_tutorials', '-DCMAKE_INSTALL_PREFIX=/home/alex/catkin_ws/install_isolated', '-G', 'Unix Makefiles']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/alex/catkin_ws/build_isolated/beginner_tutorials && cmake /home/alex/catkin_ws/src/beginner_tutorials -DCATKIN_DEVEL_PREFIX=/home/alex/catkin_ws/devel_isolated/beginner_tutorials -DCMAKE_INSTALL_PREFIX=/home/alex/catkin_ws/install_isolated -G 'Unix Makefiles'

Command failed, exiting.
`
===================== output code ^^^

Not sure where to go from here. I saw a [ROS wiki on overlaying workspaces](http://wiki.ros.org/catkin/Tutorials/workspace_overlaying) but I'm not sure if this is something I'm still able to do or not. It also seems a bit over my head so I wanted to ask here first. End goal is to be able to have everything in one workspace. 

Thank you!

Checklist:
1. Result of `rosbag_validate`:
https://gist.github.com/apletta/4ca18fad1663e4bfb4c0cd4570af9d6c

2. post a link to a Git repository containing a branch of
   `cartographer_ros` containing all the configuration, launch, and URDF files
   required to reproduce your issue - Don't know how to upload workspace to git as there are other git repo's in it.

3. Issue reproduction:
cd /home/alex/catkin_ws/build_isolated/beginner_tutorials && cmake /home/alex/catkin_ws/src/beginner_tutorials -DCATKIN_DEVEL_PREFIX=/home/alex/catkin_ws/devel_isolated/beginner_tutorials -DCMAKE_INSTALL_PREFIX=/home/alex/catkin_ws/install_isolated -G 'Unix Makefiles'

apletta, 2019-06-24:
Decided to remove all ROS from system and re-install everything. Now working fine. Closing issue.

closed 2019-06-24 by apletta

alprn42, 2019-07-18:
Hi,
I've removed all ROS and re-install but it did not work. I got same error again. Do you have any other idea to solve it? Thank you.

soorajanilkumar, 2020-02-10:
> Hi,
> I've removed all ROS and re-install but it did not work. I got same error again. Do you have any other idea to solve it? Thank you.

Check out catkin workspace chaining for more information. It should give you an idea on how to fix your issue. 
<http://wiki.ros.org/catkin/Tutorials/workspace_overlaying>