cartographer-project/cartographer_ros #1315: "Submaps: For frame [map]: Frame [map] does not exist" Error

hachbani2019, 2019-06-12:
Hi,
I've been trying to set up Cartographer on ROS. I'm going to give as much details as I can. 
I want to run cartographer on my own bag.
When I launch cartographer, I get nothing on the Rviz display, it gives me a error (screenshot: https://imgur.com/a/tq3G3jR) 

When I run rqt_graph : https://imgur.com/a/ShktkBL

Here's my rqt_tf_tree: https://imgur.com/a/ReV7MYc

I launch cartographer using this command: 
`roslaunch cartographer_ros demo_my_robot.launch bag_filename:=/home/hamza/Documents/bags_ouster/2019-05-10-11-00-03.bag`

Here's my launch file : (demo_my_robot.launch)
```
<launch>
  <param name="/use_sim_time" value="true" />

  <include file="$(find cartographer_ros)/launch/backpack_3d.launch" />
  
  <remap from="/os1_cloud_node/points" to="/points2"/>
  <node pkg="tf" type="static_transform_publisher" name="tf_pub" args="0 0 0 0 0 0 /base_link /os1_sensor 100"/>
  <node name="rviz" pkg="rviz" type="rviz" required="true"
      args="-d $(find cartographer_ros)/configuration_files/demo_3d.rviz" />
  <node name="playbag" pkg="rosbag" type="play"
      args="--clock $(arg bag_filename)" />
</launch>
```
My_robot.lua : 

```
options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false,
  use_odometry = false,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 0,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 64,
  num_point_clouds = 1,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 160

MAP_BUILDER.use_trajectory_builder_3d = true
MAP_BUILDER.num_background_threads = 7
POSE_GRAPH.optimization_problem.huber_scale = 5e2
POSE_GRAPH.optimize_every_n_nodes = 320
POSE_GRAPH.constraint_builder.sampling_ratio = 0.03
POSE_GRAPH.optimization_problem.ceres_solver_options.max_num_iterations = 10
POSE_GRAPH.constraint_builder.min_score = 0.62
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.66

return options
```
I've been trying to figure out the problem, anyone can help please ?

omair-khalid, 2019-06-12:
In your rqt graph, it can be seen that the cartographer node is not connected to the /points2 topic. Make sure that the launch file (backpack_3d.launch) is pointing to your My_robot.lua config file.

I am not sure about this but you can try  to exclude the "/" before the "os1" in <remap from="points2" to="/os1_cloud_node/points"/>.

hachbani2019, 2019-06-12:
Thanks @omair-khalid for you answer. As a ROS newbie, I don't know how to point the .launch to the my_robot.lua Can you show me how please ? couldn't find the answer online. 

Since you brought it up, do you know whether the right syntax is "remap from="points2" to="/os1_cloud_node/points"/." or "remap from="/os1_cloud_node/points" to="points2"/." ? I find different answers online. Did try removing the "/" though, didn't change anything.

hachbani2019, 2019-06-14:
Hi @omair-khalid , I did figure out how to point the .launch to my_robot.lua, It was already pointing to it. I did try your suggesion as well about the remap command, nothing changed. Do you have any other idea how to fix this please ? 
Thanks

omair-khalid, 2019-06-14:
Perhaps, you also need to define the transformation between the os1_sensor frame to os1_lidar frame. This would be required if your lidar data is published in the os1_lidar frame. 

Check the message type of your data. Is sensor_msgs/LaserScan or sensor_msgs/PointCloud2? I bring this up because I am also using a 2dLidar (from OMRON) and my data is of the sensor_msgs/LaserScan and I set  num_laser_scans = 1 and num_point_clouds =0 in the lua file.

For the remapping command syntax, this is working for me :
`  <node name="cartographer_node" pkg="cartographer_ros" type="cartographer_node" args="-configuration_directory $(find cartographer_ros)/configuration_files -configuration_basename my_robot.lua" output=      "screen">
    <remap from="odom" to="robot_pose_ekf/odom_combined_ekf" />     
    <remap from="imu" to="imu/data"/>
  </node>`

hachbani2019, 2019-06-14:
my message's type is sensors_msgs/PointCloud2. IS this how I can define the ransformation you did talk about ? 
  <node pkg="tf" type="static_transform_publisher" name="tf_pub" args="0 0 0 0 0 0 /os1_lidar /os1_sensor 100"/>

omair-khalid, 2019-06-14:
Okay, sorry, I was mistaken about the os1_sensor frame to os1_lidar frame transformation issue. I can now see from your tf tree that it already exists.

If possible, could you provide a link to your bag file so that someone else or I can try running it?

ayu135, 2019-06-14:
I am facing a similar issue too. I am using a 2D lidar with messages published on /scan topic from a bag and the tf for the sensor to base link are generated from the urdf using robot state publisher. I am getting the same output in rviz as well.

hachbani2019, 2019-06-14:
@omair-khalid My bag file is 6.7Gb large, I'll try to upload it this weekend for my fiber optic network at home. Do you happen to know anyway where I can upload large files ? 

@ayu135 wish I could help, can you put the link of your issue  ?

ayu135, 2019-06-18:
@hachbani2019 I haven't posted my issue separately I am having the exact same issue you are having. Did you manage to solve it?

MichaelGrupp, 2020-10-23:
Closing as inactive.

closed 2020-10-23 by MichaelGrupp