cartographer-project/cartographer_ros #1316: Start new trajectory with master branch

francescoricciuti, 2019-06-12:
Hi everybody,

I'm using the master branch of Cartographer, and I need to finish a trajectory and start a new one.

In Release 1.0 I was doing it by finishing the trajectory with the service 
`rosservice call /finish_trajectory 1`

and calling the `cartographer_start_trajectory` node

`rosrun cartographer_ros cartographer_start_trajectory ... etc.`

The problem is that the `cartographer_start_trajectory` node is not present anymore in the master branch.

Did the name of the node change? Was the logic moved to another node?

Thanks in advance.

Francesco

MichaelGrupp, 2019-06-13:
This executable was removed in favor of a simplified service.

Just use
`rosservice call /start_trajectory ...` (tab complete for all options)

or using ROS service API from within your code.

PRs:
* API change: https://github.com/googlecartographer/cartographer_ros/pull/1245
* documentation(open): https://github.com/googlecartographer/cartographer_ros/pull/1282

Background: https://github.com/googlecartographer/rfcs/blob/master/text/0028-simplify-ros-start-trajectory.md

francescoricciuti, 2019-06-13:
Great!

Thank you very much!

closed 2019-06-13 by francescoricciuti