cartographer-project/cartographer_ros #1319: Follow ROS security updates & remove EOL distro Indigo

MichaelGrupp, 2019-06-13:
GPG keys of pakages.ros.org have changed because of a security issue:
https://discourse.ros.org/t/security-issue-on-ros-build-farm/9342
https://discourse.ros.org/t/new-gpg-keys-deployed-for-packages-ros-org/9454

The [melodic base image of cartographer_ros](https://github.com/googlecartographer/cartographer_ros/blob/master/Dockerfile.base.melodic) has the outdated key. This leads to problems while updating rosdeps (if no caching is used), see this CI log of our dummy repository:
```
Err:2 http://packages.ros.org/ros/ubuntu bionic InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
```

---

Additionally, ROS Indigo has reached EOL and can be removed from CI builds. They are currently broken because of that.

---

- [ ] re-build & push base image (pulls new image with [fresh keys](https://discourse.ros.org/t/new-gpg-keys-deployed-for-packages-ros-org/9454/3))
- [ ] remove Indigo CI

bochen87, 2020-06-16:
@MichaelGrupp I just noticed CI is broken since merge of #1379 
Do you know who has access to the melodic base image? It seems to be stored on the Google Cloud Platform

bochen87, 2020-06-17:
should be fixed by #1484

closed 2020-06-17 by MichaelGrupp