cartographer-project/cartographer_ros #1321: How to use start_trajectory

carl1218, 2019-06-13:
Hi,
  When i finished the trajectory,I wan to start the new trajectory,i runned  `rosservice call /start_trajectory`,but can not start the new trajectory.
 How to use the start_trajectory?What commands should I call?

Thank you

narogon, 2019-06-17:
you must finish the old trajectory first. 

`rosservice call /finish_trajectory "trajectory_id: 1"`
`rosrun cartographer_ros cartographer_start_trajectory  -configuration_directory PATH OF CONF FOLDER -configuration_basename NAME OF LUA FILE -initial_pose '{to_trajectory_id = 0, timestamp=0,relative_pose = { translation = { 0.0, 0., 0. }, rotation = { 0.0, 0.0, 0.0 } } }'`

carl1218, 2019-06-20:
@narogon 
 Thanks for your prompt reply.
I was doing it by finishing the trajectory with the service
`rosservice call /finish_trajectory 1`and calling the cartographer_start_trajectory node
`rosrun cartographer_ros cartographer_start_trajectory ... etc.`
But,the cartographer_start_trajectory node is not present anymore in the master branch.the name of the node changed.

cassieqiuyd, 2019-06-28:
If you use the newest version of cartographer_ros on GitHub, they actually simplified the arguments required for start_trajectory.  You can use 

`rosservice call /start_trajectory "configuration_directory: ''
configuration_basename: ''
use_initial_pose: false
initial_pose:
  position: {x: 0.0, y: 0.0, z: 0.0}
  orientation: {x: 0.0, y: 0.0, z: 0.0, w: 0.0}
relative_to_trajectory_id: 0" 
`
one trick is to type `rosservice call /start_trajectory` in the command line and press `tab`. and it will show the corresponding arguments needed.

closed 2020-10-14 by MichaelGrupp