cartographer-project/cartographer_ros #1322: My pure location model cannot work well

Zacharyzqc, 2019-06-14:
1. run `rosbag_validate`  
 and got the message 'Point data (frame_id: "rslidar") has a large gap, largest is 0.102116 s, recommended is [0.0005, 0.05] s with no jitter'. How can I improve it
2. the setting files link : [files](https://github.com/Zacharyzqc/cartographer-settings)
3.the pic,we can see that the lidar points cannot match the map suitable
[image]

CherryChenw, 2019-07-23:
mybag is the same error to you，and I can't see anything when I running with rslidar and imu。
[ WARN] [1563854413.115772888]: W0723 12:00:13.000000  5669 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)
 Have you seen the same question？

MichaelGrupp, 2020-10-23:
Closing as inactive. We can't help with tuning this custom setup unfortunately.

closed 2020-10-23 by MichaelGrupp