cartographer-project/cartographer_ros #1323: E: There were unauthenticated packages and -y was used without --allow-unauthenticated ERROR: the following rosdeps failed to install   apt: command [apt-get install -y ros-kinetic-pcl-conversions] failed

mojovski, 2019-06-14:
After calling
```
docker build --network=host -t cartographer_kinetic -f Dockerfile.kinetic .
```
I receive
```
WARNING: The following packages cannot be authenticated!
  libvtk6.2 libvtk6.2-qt libvtk6-java python-vtk6 tcl-vtk6 vtk6 libvtk6-dev
  libvtk6-qt-dev ros-kinetic-pcl-msgs ros-kinetic-pcl-conversions
E: There were unauthenticated packages and -y was used without --allow-unauthenticated
ERROR: the following rosdeps failed to install
  apt: command [apt-get install -y ros-kinetic-pcl-conversions] failed
executing command [apt-get install -y ros-kinetic-pcl-conversions]
The command '/bin/sh -c cartographer_ros/scripts/install_debs.sh && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 1

```
coming from this line: 
https://github.com/googlecartographer/cartographer_ros/blob/master/scripts/install_debs.sh#L44

MichaelGrupp, 2019-06-14:
Probably related to https://github.com/googlecartographer/cartographer_ros/issues/1319

mojovski, 2019-09-04:
@MichaelGrupp  thanks!
In short:
```
sudo apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
sudo apt-key del 421C365BD9FF1F717815A3895523BAEEB01FA116
```
helps. See https://discourse.ros.org/t/new-gpg-keys-deployed-for-packages-ros-org/9454

closed 2019-09-04 by mojovski