cartographer-project/cartographer_ros #1326: Ignored subdivision

MrGitGo, 2019-06-18:
I have set up google cartographer many times for 2d and 3d. But now I wanted to use a rgbd camera for 2d mapping. In order to do this I converted the point cloud to laserscan, with this package: http://wiki.ros.org/pointcloud_to_laserscan

I use a pico monstar.

When I launch google cartographer with the pc_to_laserscan I get this warning:

`[ WARN] [1560845643.823460226]: W0618 10:14:03.000000 18489 sensor_bridge.cc:200] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 636964424437691470 is not before current subdivision time 636964424437691470`

fengfanrop, 2020-03-09:
@MrGitGo I met the same problem with you ,have you solved it out?

Gerdid, 2020-06-26:
I also have the same problem, any idea?

MrGitGo, 2020-06-28:
I never got a satisfactory result. Try it with laser sensors instead

Gerdid, 2020-06-28:
I was using laser scan. I think this problem is caused by the stamp part of the message

MichaelGrupp, 2020-10-16:
This can happen sometimes during scan unskewing. If it doesn't affect SLAM performance, you can usually ignore sporadic warnings. Besides that, it can also indicate timing issues.

closed 2020-10-16 by MichaelGrupp

MichaelGrupp, 2020-11-09:
The problem is: `time_increment: 0.0`. This needs to be set to the correct value, otherwise Cartographer's scan unskewing doesn't work correctly. See also http://docs.ros.org/en/api/sensor_msgs/html/msg/LaserScan.html

Awacky, 2022-04-15:
> 我已经为 2d 和 3d 多次设置了 google cartographer。但现在我想使用 rgbd 相机进行 2d 映射。为了做到这一点，我将点云转换为laserscan，使用这个包：[http ://wiki.ros.org/pointcloud_to_laserscan](http://wiki.ros.org/pointcloud_to_laserscan)
> 
> 我用的是pico monstar。
> 
> 当我使用 pc_to_laserscan 启动 google cartographer 时，我收到以下警告：
> 
> `[ WARN] [1560845643.823460226]: W0618 10:14:03.000000 18489 sensor_bridge.cc:200] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 636964424437691470 is not before current subdivision time 636964424437691470`
> 
> 这是 /scan 的标题：
> 
> ```
>   seq: 100
>   stamp: 
>     secs: 1560845618
>     nsecs: 155853000
>   frame_id: "royale_camera_optical_frame"
> angle_min: -3.1400001049
> angle_max: 3.1400001049
> angle_increment: 0.00870000012219
> time_increment: 0.0
> scan_time: 0.333299994469
> range_min: 0.449999988079
> range_max: 10.0
> ranges: [inf, inf, inf, 0.45439407229423523, 0.45167145133018494, inf, inf,
> ```

 I met the same problem with you ,have you solved it out?