cartographer-project/cartographer_ros #1328: Why is there no landmark constraint?

GeesonCheung, 2019-06-19:
Hello,
I have a new understanding of landmark based on the above information. Recently, I have been studying how to better use landmark.Now I run the script file provided by @MichaelGrupp to publish landmark with an empty landmark with an interval of 0.1s and a sample landmark with an interval of 5s.Althoght landmark was displayed on the map when building the map,landmark‘s constraints were not seen, all constraints were generated by scan and submap, which was inconsistent with video display provided by @pifon2a . All constraints on video were established with landmark.

In my opinion, landmark must have constraints in order for it to work. I beg your help.

Here are my lua files:

'''
include"map_builder.lua"
include "trajectory_builder.lua"

options = {
map_builder = MAP_BUILDER,
trajectory_builder = TRAJECTORY_BUILDER,
map_frame = "map",
tracking_frame = "i_imu",
published_frame = "i_base_link",
odom_frame = "odom",
provide_odom_frame = false,
publish_frame_projected_to_2d = false,
use_odometry = true,
use_nav_sat = false,
use_landmarks = true,
num_laser_scans = 1,
num_multi_echo_laser_scans = 0,
num_subdivisions_per_laser_scan = 1,
num_point_clouds = 0,
lookup_transform_timeout_sec = 0.2,
submap_publish_period_sec = 0.3,
pose_publish_period_sec = 5e-3,
trajectory_publish_period_sec = 30e-3,
rangefinder_sampling_ratio = 1.,
odometry_sampling_ratio = 1.,
fixed_frame_pose_sampling_ratio = 1.,
imu_sampling_ratio = 1.,
landmarks_sampling_ratio = 1.,
}

TRAJECTORY_BUILDER_2D.num_accumulated_range_data = 1
MAP_BUILDER.use_trajectory_builder_2d = true

TRAJECTORY_BUILDER_2D.submaps.num_range_data = 35--35
TRAJECTORY_BUILDER_2D.min_range = 0.1
TRAJECTORY_BUILDER_2D.max_range = 30.
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 1.
TRAJECTORY_BUILDER_2D.use_imu_data = true

TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.linear_search_window = 0.1
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.translation_delta_cost_weight = 10.
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.rotation_delta_cost_weight = 1e-1
POSE_GRAPH.constraint_builder.min_score = 0.4--0.55 0.4
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.5 --0.8 0.5
POSE_GRAPH.optimization_problem.huber_scale = 1e2
POSE_GRAPH.optimize_every_n_nodes = 35--35
POSE_GRAPH.constraint_builder.min_score = 0.65

return options
'''
[image]
[image]

JaredLee1218, 2019-07-19:
I think you need to make subscribe of landmakrs

could you tell me how you made landmarks? T^T

MichaelGrupp, 2020-10-16:
You can try this testing script to publish fake landmarks: https://github.com/cartographer-project/cartographer_ros/blob/master/cartographer_ros/scripts/dev/publish_fake_random_landmarks.py

closed 2020-10-16 by MichaelGrupp

CumaOzavci, 2021-01-12:
I have the same "problem". But i don't think the lines in the video are constraints. They seem more like just landmark observations. Can @pifon2a clarify here?

And if those lines are just observations, how can i ensure cartographer is processing landmarks correctly? Is debugging /landmark_poses_list topic enough?