cartographer-project/cartographer_ros #1330: no transform from map to base_link

Colidar, 2019-06-24:
hi there,

i am using real time 2D cartographer with a RPLidar A2 and IMU MPU9250 on ROS melodic.
The IMU is connected to a arduino mega and publishes sensor_msgs/Imu over rosserial. The arduino sketch also publishes tf data (base_link to imu_link).

After i wanted to integrate imu to cartographer. In my lua file i changed:
tracking_frame: imu_link
published_frame: base_link

I have no odometry data, that's why :
provide_odom_frame: false

And i have also created a static transform publisher in my launch file to publish tf from base_link to laser_link:
`<node pkg="tf" type="static_transform_publisher" name="base_to_laser" 
    args="0 0 0 0 0 0 base_link laser_link 100" />` 

When I am starting cartographer with:
`roslaunch cartographer_ros demo_my_robot.launch`
there is no map shown in rviz (Error fixed frame map doesn't exist)

I checkt tf and there is only the transformation from base_link to imu_link:
https://www.dropbox.com/s/tyan9mb4cc00rxv/frames.pdf?dl=0

Does anyone know,  why cartographer is not publishing anything anymore? And why there is no tf from map to base_link??

My lua, launch, urdf and ino files are on dropbox:
https://www.dropbox.com/sh/nhamff0hnx3jg87/AABljxW8ZzjPKvcs2eOWSzbUa?dl=0

LostXine, 2019-06-27:
hi @Colidar 
When cartographer didn't get data from sensors on time, it usually would publish nothing. You can add `output="screen"` to the cartographer node in your launch file and check the console. It would be tons of ROS_WARN like still waiting for XX msgs or anything similar. Please double check your topic publishers and subscribers with `rqt_graph` or `rostopic`.
And one more thing, I noticed that you set `TRAJECTORY_BUILDER_2D.use_imu_data = false` in your lua file. According to my knowledge, it should be `true` if you want cartographer subscribe imu topic.
Hope it helps, good luck.

Colidar, 2019-07-02:
@LostXine 
Thanks for your answer! You were right, i set `TRAJECTORY_BUILDER_2D.use_imu_data = true` and i had to set `/use_sim_timevalue="false"`
Now it's working

closed 2019-07-02 by Colidar