cartographer-project/cartographer_ros #1334: Dockerfile for ros melodic is broken

razorheadfx, 2019-06-25:
Trying to build the current Dockerfile.melodic (at #92a8b81) like so:```docker build -t cartographer_broken -f Dockerfile.melodic .``` breaks.
```
Sending build context to Docker daemon  3.761MB
Step 1/25 : FROM eu.gcr.io/cartographer-141408/cartographer_ros_melodic_base
 ---> 4494cdd37eac
Step 2/25 : ARG CARTOGRAPHER_VERSION=master
 ---> Using cache
 ---> 12fba05dfdf3
Step 3/25 : ARG github_token
 ---> Using cache
 ---> 91a30d1d3d94
Step 4/25 : RUN apt-get update && apt-get install -y sudo && rm -rf /var/lib/apt/lists/*
 ---> Running in 86ce8532ce8f
Get:1 http://archive.ubuntu.com/ubuntu bionic InRelease [242 kB]
Get:2 http://archive.ubuntu.com/ubuntu bionic-updates InRelease [88.7 kB]
Get:3 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]
Get:4 http://archive.ubuntu.com/ubuntu bionic-backports InRelease [74.6 kB]
Get:5 http://packages.ros.org/ros/ubuntu bionic InRelease [4669 B]
Get:6 http://archive.ubuntu.com/ubuntu bionic/universe Sources [11.5 MB]
Err:5 http://packages.ros.org/ros/ubuntu bionic InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
Get:7 http://archive.ubuntu.com/ubuntu bionic/main amd64 Packages [1344 kB]
Get:8 http://archive.ubuntu.com/ubuntu bionic/restricted amd64 Packages [13.5 kB]
Get:9 http://archive.ubuntu.com/ubuntu bionic/universe amd64 Packages [11.3 MB]
Get:10 http://security.ubuntu.com/ubuntu bionic-security/universe Sources [188 kB]
Get:11 http://archive.ubuntu.com/ubuntu bionic/multiverse amd64 Packages [186 kB]
Get:12 http://archive.ubuntu.com/ubuntu bionic-updates/universe Sources [325 kB]
Get:13 http://archive.ubuntu.com/ubuntu bionic-updates/universe amd64 Packages [1226 kB]
Get:14 http://archive.ubuntu.com/ubuntu bionic-updates/multiverse amd64 Packages [7239 B]
Get:15 http://archive.ubuntu.com/ubuntu bionic-updates/restricted amd64 Packages [10.8 kB]
Get:16 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 Packages [862 kB]
Get:17 http://archive.ubuntu.com/ubuntu bionic-backports/main amd64 Packages [2496 B]
Get:18 http://archive.ubuntu.com/ubuntu bionic-backports/universe amd64 Packages [3927 B]
Get:19 http://security.ubuntu.com/ubuntu bionic-security/restricted amd64 Packages [5436 B]
Get:20 http://security.ubuntu.com/ubuntu bionic-security/universe amd64 Packages [720 kB]
Get:21 http://security.ubuntu.com/ubuntu bionic-security/main amd64 Packages [558 kB]
Get:22 http://security.ubuntu.com/ubuntu bionic-security/multiverse amd64 Packages [4169 B]
Reading package lists...
W: GPG error: http://packages.ros.org/ros/ubuntu bionic InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
E: The repository 'http://packages.ros.org/ros/ubuntu bionic InRelease' is not signed.
The command '/bin/sh -c apt-get update && apt-get install -y sudo && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 100

```

Most likely due to new GPG key deployed for packages.ros.org [in the wake of a ros build farm compromise](https://discourse.ros.org/t/new-gpg-keys-deployed-for-packages-ros-org/9454).

Also: unlike the other Dockerfiles ```Dockerfile.melodic``` does not use the official ros image but something else (which is probably why it broke in this way)
https://github.com/googlecartographer/cartographer_ros/blob/92a8b81a8489dd71a73c6d838c560712311b1da5/Dockerfile.melodic#L14-L16
 ```Dockerfile.lunar``` for reference
https://github.com/googlecartographer/cartographer_ros/blob/92a8b81a8489dd71a73c6d838c560712311b1da5/Dockerfile.lunar#L14-L16

razorheadfx, 2019-06-25:
Update: apparently there is another issue with this Dockerfile
Deriving from ```ros:melodic``` and trying to build again breaks with:
```
...
Step 8/25 : RUN CARTOGRAPHER_VERSION=$CARTOGRAPHER_VERSION     cartographer_ros/scripts/update_catkin_workspace.sh
 ---> Running in ae006a891d4f

. /opt/ros/${ROS_DISTRO}/setup.sh
#!/usr/bin/env sh
# generated from catkin/cmake/template/setup.sh.in

# Sets various environment variables and sources additional environment hooks.
# It tries it's best to undo changes from a previously sourced setup file before.
# Supported command line options:
# --extend: skips the undoing of changes from a previously sourced setup file
# --local: only considers this workspace but not the chained ones
# In plain sh shell which doesn't support arguments for sourced scripts you can
# set the environment variable `CATKIN_SETUP_UTIL_ARGS=--extend/--local` instead.

# since this file is sourced either use the provided _CATKIN_SETUP_DIR
# or fall back to the destination set at configure time
: ${_CATKIN_SETUP_DIR:=/opt/ros/melodic}
_SETUP_UTIL="$_CATKIN_SETUP_DIR/_setup_util.py"
unset _CATKIN_SETUP_DIR

if [ ! -f "$_SETUP_UTIL" ]; then
  echo "Missing Python script: $_SETUP_UTIL"
  return 22
fi

# detect if running on Darwin platform
_UNAME=`uname -s`
_IS_DARWIN=0
if [ "$_UNAME" = "Darwin" ]; then
  _IS_DARWIN=1
fi
unset _UNAME

# make sure to export all environment variables
export CMAKE_PREFIX_PATH
if [ $_IS_DARWIN -eq 0 ]; then
  export LD_LIBRARY_PATH
else
  export DYLD_LIBRARY_PATH
fi
unset _IS_DARWIN
export PATH
export PKG_CONFIG_PATH
export PYTHONPATH

# remember type of shell if not already set
if [ -z "$CATKIN_SHELL" ]; then
  CATKIN_SHELL=sh
fi

# invoke Python script to generate necessary exports of environment variables
# use TMPDIR if it exists, otherwise fall back to /tmp
if [ -d "${TMPDIR:-}" ]; then
  _TMPDIR="${TMPDIR}"
else
  _TMPDIR=/tmp
fi
_SETUP_TMP=`mktemp "${_TMPDIR}/setup.sh.XXXXXXXXXX"`
unset _TMPDIR
if [ $? -ne 0 -o ! -f "$_SETUP_TMP" ]; then
  echo "Could not create temporary file: $_SETUP_TMP"
  return 1
fi
CATKIN_SHELL=$CATKIN_SHELL "$_SETUP_UTIL" $@ ${CATKIN_SETUP_UTIL_ARGS:-} >> "$_SETUP_TMP"
_RC=$?
[... 115 lines cut ...]
PYTHON_CODE_BUILD_ROSLISP_PACKAGE_DIRECTORIES=$(cat <<EOF
from __future__ import print_function
import os
env_name = 'CMAKE_PREFIX_PATH'
paths = [path for path in os.environ[env_name].split(os.pathsep)] if env_name in os.environ and os.environ[env_name] != '' else []
workspaces = [path for path in paths if os.path.exists(os.path.join(path, '.catkin'))]
paths = []
for workspace in workspaces:
    filename = os.path.join(workspace, '.catkin')
    data = ''
    with open(filename) as f:
        data = f.read()
    if data:
        paths.append(os.path.join(workspace, 'share', 'common-lisp'))
print(os.pathsep.join(paths))
EOF
)
export ROSLISP_PACKAGE_DIRECTORIES="`python -c \"$PYTHON_CODE_BUILD_ROSLISP_PACKAGE_DIRECTORIES\"`"
unset _i

unset _CATKIN_ENVIRONMENT_HOOKS_COUNT

cd catkin_ws/src
cartographer_ros/scripts/update_catkin_workspace.sh: 22: cd: can't cd to catkin_ws/src
```

MichaelGrupp, 2019-07-12:
Looks like https://github.com/googlecartographer/cartographer_ros/issues/1319

MichaelGrupp, 2019-07-12:
Docker only uses a different base image with pre-built dependencies because there were issues with the CI pipeline a few months ago. Therefore current `Dockerfile.melodic` can't be built directly `FROM ros:melodic`. You can however change it to build directly from the `ros:melodic` base again, this would be essentially reverting this patch: https://github.com/googlecartographer/cartographer_ros/pull/1021 

You could also build `Dockerfile.melodic.base` yourself and just change the `FROM` statement in `Dockerfile.melodic` to include your freshly built base image instead of the outdated one from Google.

bochen87, 2020-06-17:
should be fixed by #1484

closed 2020-06-17 by bochen87