cartographer-project/cartographer_ros #1338: Wall disappears in 2D SLAM

wkwkw, 2019-06-27:
I'm trying cartographer's 2D SLAM and met the problem that the wall, which once has been recognized, be disappeared when the robot returns.

It had not been happend with Gmapping, so I want the cartographer's result to be the same level.

* cartographer
[image]
* gmapping
[image]
* photo of the area, taken from the entrance, left side in the upper images (the way is about 1m wide): 
[image]

How can I fix this?

I think the wall's smooth surface is causing problem (It may be hard to recognize for laser sensor).
But I believe the effect can be mitigated in software side.

Sensors: 
* 2D laser sensor (/scan)
* wheel odometry (/odom)
* IMU (/imu)

[forked repo](https://github.com/wkwkw/cartographer_ros.git)
[bagfile](https://github.com/wkwkw/tmp/blob/master/carto_files/locker_area.bag)
[result of rosbag_validate](https://gist.github.com/wkwkw/bd16c68e35ebfd9fceca6fe474debaec)

Thanks in advance for your help!

MichaelGrupp, 2019-07-05:
I'd look at the submaps display instead of occupancy grid, The occupancy grid node of Cartographer just blends all submaps into an image, so it could be that it just _looks_ as if the wall is gone.

The submaps plugin of cartographer_rviz is able to show you the individual submaps instead, which might be better for checking the result here. Note that you also have similar laser artifacts ("spray" of free space) in the gmapping result, maybe because of the reflective (?) surface of your walls.

wkwkw, 2019-07-08:
Thanks for the response.

First, I checked the individual submaps on rviz and found that the first submap, no 0, still holds the wall and others don't. I show the all submaps below (total 6 with default _num_range_data_ =90).

Is there a way to use the better result of the submap in navigating the robot? (I'm considering to use the Navigation Stack)

0 [image]
1 [image]
2 [image]
3 [image]
4 [image]
5 [image]

Second, as you mentioned, "spray" of free space can also be found in gmapping result (and also in the result of official data checker tool of HOKUYO) and I think it's caused by the reflective surface. 

I guess gmapping tends to keep what is recognized to be a wall as a wall and cartographer not.
It seems to be a tradeoff between erasing what have to be kept and failing to erase what have to be erased.

p.s.
I found that setting _max_range_ to around 5m reduces the probrem, but I want to utilize the 20m view range of the sensor (HOKUYO UST-20LX-H01).

Yu-Xiaoxian, 2019-08-08:
@wkwkw for the reflective surface, I suggest to filter some points based on their intensity.

Yu-Xiaoxian, 2019-08-08:
@wkwkw I am interested to hear that you are using Hokuyo lidar, have you ever confused by the problem that this lidar get inaccurate at near distance(0.2m - 1.0m)?

wkwkw, 2019-08-09:
@Yu-Xiaoxian Hi, I've just been trying the way you mentioned and getting better results.
I'm now feeling more confident in taking this way by your reply, thank you.
> @wkwkw for the reflective surface, I suggest to filter some points based on their intensity.

And for the question about Hokuyo lidar, no. I've never been met a problem by the inaccurateness at near distance so far.

Yu-Xiaoxian, 2019-08-13:
@wkwkw great, I'm glad to hear you making progress. Thanks for your info about the Hokuyo Lidar, I will contact my supplier for a solution.

closed 2020-10-16 by MichaelGrupp