cartographer-project/cartographer_ros #1340: Obtaining trajectory markers

alankdkd, 2019-06-27:
Hello,

I'm trying to get poses (position + orientation) along a trajectory.  In #332, SirVer says "Btw, Cartographer now also publishes the trajectory in form of markers, so you can get to the information now."  But after much searching, I don't see how to do this.

My guess is that I need to include another node that subscribes to the published markers.  I have two questions: 1) Do markers have orientation in addition to positions?  and 2) How can I get these markers?  

It would be great if there's an existing node that subscribes to markers that I could modify.

Thanks!
Alan

P.S. This is a general question not specific to any particular data or scenario.

ojura, 2019-06-28:
I believe there is a service that allows you to query cartographer for a trajectory.

alankdkd, 2019-07-01:
Thanks ojura.  In case anyone else is looking for this info, here are steps to obtain it:

1) The Cartographer node publishes topic /trajectory_node_list, 
2) This topic is of type visualization_msgs/MarkerArray, 
3) Its message contains an array of Markers, 
4) Markers contain a member of type geometry_msgs/Pose, 
5) the Pose type contains two fields: position and orientation, 
6) Pose.position is three floats and Pose.orientation is a quaternion (four floats).  
7) Add (or modify) a node to subscribe to the /trajectory_node_list, and write the Poses to a file.

alankdkd, 2019-07-01:
BTW, I haven't implemented these steps in code yet.  I'll post an update when I have more information.

ayu135, 2019-07-02:
I just implemented it using the steps you mentioned above and it does not work for me. The output of the marker array pose type does not contain the actual pose. The output is the following:
```
 header: 
      seq: 0
      stamp: 
        secs: 1561636493
        nsecs: 179813088
      frame_id: "map"
    ns: "Trajectory 0"
    id: 2
    type: 4
    action: 0
    pose: 
      position: 
        x: 0.0
        y: 0.0
        z: 0.05
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 1.0
```
The actual trajectory is present in the points field in the message as only x,y and z values without orientation
```
points: 
      - 
        x: 0.863043830642
        y: 6.99164155632
        z: 0.0
      - 
        x: 0.860062564869
        y: 6.93523299369
        z: 0.0
      - 
        x: 0.858788512093
        y: 6.88456487155
        z: 0.0
      - 
        x: 0.867231202534
        y: 6.81827431059
        z: 0.0
      - 
        x: 0.866676887132
        y: 6.75207955557
        z: 0.0
      - 
```

alankdkd, 2019-07-03:
@ayu135 if you post your code, I'll take a look at it.  (Won't be in tomorrow; U.S. holiday)

alankdkd, 2019-07-08:
@ayu135 I'm getting the same result.  Looking into it.  I'll post if/when I find a solution.

ojura, 2019-07-10:
As I have said, there is a service for querying Cartographer for full trajectory poses, not just positions. It is called `TrajectoryQuery`.

alankdkd, 2019-07-10:
@ayu135: The reason the previous attempts failed is the frame of reference is ON the robot, so doesn't change as the robot moves.  The solution is to adjust the position and orientation by the transformation from the /map frame to the /base_link frame.

The complete solution with source, Package.xml, and CMakeLists.txt is in item #1350.

closed 2019-07-10 by alankdkd