cartographer-project/cartographer_ros #1343: How to choose tracking_frame, published_frame... ?

hachbani2019, 2019-06-28:
Hi, 

I've been struggling to setup Cartographer for quite a while now. I came to the point when I launch my command `roslaunch cartographer_ros demo_my_robot.launch bag_filename:=my_bagfile.bag`

It gives me the following error message: 
```
[ WARN] [1561711713.440545548, 1557479058.814241016]: W0628 10:48:33.000000 24764 tf_bridge.cc:52] Lookup would require extrapolation into the past.  Requested time 328.765268900 but the earliest data is at time 1557479048.905276052, when looking up transform from frame [base_link] to frame [os1_imu]
[ WARN] [1561711713.445208674, 1557479058.824302278]: W0628 10:48:33.000000 24764 tf_bridge.cc:52] Lookup would require extrapolation into the past.  Requested time 328.765268900 but the earliest data is at time 1557479048.905276052, when looking up transform from frame [base_link] to frame [os1_imu]
```

### Before going into details, here's my question : 

How do we chosse the right map_frame, tracking_frame, published_frame, and the fixed_frame on Rviz ? 

### **backpack_3d.launch**
```
<launch>
  <param name="robot_description"
    textfile="$(find cartographer_ros)/urdf/backpack_3d.urdf" />

  <node name="robot_state_publisher" pkg="robot_state_publisher"
    type="robot_state_publisher" />

  <node pkg="tf" type="static_transform_publisher" name="tf_pub" args="0 0 0 0 0 0 /base_link /os1_sensor 100"/>

  <node name="cartographer_node" pkg="cartographer_ros"###
o
      type="cartographer_node" args="
          -configuration_directory $###(find cartographer_ros)/configuration_files
          -configuration_basename backpack_3d.lua"
      output="screen">
    <remap from="points2" to="/os1_cloud_node/points" />
    <remap from="imu" to="/os1_cloud_node/imu" />
  </node>

  <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
      type="cartographer_occupancy_grid_node" args="-resolution 0.05" />
      
</launch>
```
### ** backpack_3d.lua :** 
```
options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "os1_imu",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false ,
  use_odometry = false,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 0,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 1,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 160

MAP_BUILDER.use_trajectory_builder_3d = true
MAP_BUILDER.num_background_threads = 7
POSE_GRAPH.optimization_problem.huber_scale = 5e2
POSE_GRAPH.optimize_every_n_nodes = 320
POSE_GRAPH.constraint_builder.sampling_ratio = 0.03
POSE_GRAPH.optimization_problem.ceres_solver_options.max_num_iterations = 10
POSE_GRAPH.constraint_builder.min_score = 0.62
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.66

return options
```
### ** my tf_tree :**  https://imgur.com/a/ReV7MYc

LostXine, 2019-07-01:
hi @hachbani2019 
According to your error message, it was the different times made some tf tree out of date. To fix it, you can add `<param name="use_sim_time" value="true"/>` to your launch file. And please make sure you really need to set "backpack_3d.urdf" as your `robot_description`.
Then it comes to the frames. You can find document [here](https://google-cartographer-ros.readthedocs.io/en/latest/configuration.html).
In short, typically `fixed_frame` on Rviz is the `map_frame` called "map". As its name shows that `map_frame` is the frame of map, and it should be the root in tf tree. `tracking_frame` is usually the frame of IMU if IMU is available, and you've done a great job.
The result of cartographer is included in the transform from `map_frame` to `published_frame`. Cartographer will always try to maintain such a transform. For example, if your robot has an odometer, you should set `published_frame` to the frame of odometer and the tf tree will look like `map_frame -(cartographer)-> odom_frame(published_frame) -(odometer node)-> base_link`. Otherwise,  you might want to choose `base_link` as your `published_frame` and your tf tree will be something more simple like  `map_frame -(cartographer)-> base_link(published_frame)`.
Finally, if `provide_odom_frame` is set, cartographer will publish tf transform from  `odom_frame` to `published_frame`. It would be very useful when the odometer node didn't publish tf message itself.
Hope it helps and good luck.

miroslavradojevic, 2020-07-04:
How should tracking_frame be set in case there is no imu recorded in the .bag file but only PointCloud2 topic?

MichaelGrupp, 2020-10-23:
> How should tracking_frame be set in case there is no imu recorded in the .bag file but only one PointCloud2 topic with laser scans?

3D mode requires an IMU.
https://google-cartographer-ros.readthedocs.io/en/latest/faq.html#why-is-imu-data-required-for-3d-slam-but-not-for-2d

closed 2020-10-23 by MichaelGrupp

genius-hl, 2021-08-22:
> hi @hachbani2019
> According to your error message, it was the different times made some tf tree out of date. To fix it, you can add `<param name="use_sim_time" value="true"/>` to your launch file. [...]

Your explanation is very clear and helpful,thanks!