cartographer-project/cartographer_ros #1344: Configuration and loop closure problems in 2d

carl1218, 2019-06-30:
Hi,
     I can't make the good result using real time processing data in 2d lidar slam.
     I only used one lidar to tried to build a map using real time, but the position not correct (when the  returned to starting point) and make cartographer failed to make a loop closure.

   A loop-closed map with significant overlap,look like this:
[image]

This is my bag:
https://drive.google.com/file/d/1QJ269tklFBGTlkQ85D_I-JMHvqe2RnUq/view?usp=sharing
This is lua:
https://drive.google.com/file/d/1UapH0d80SXHOfGMS01-KvsJCmyMspn4n/view?usp=sharing
This is launch:
https://drive.google.com/file/d/1RLAbbk1HHL2dycQ8jzbKIh5URlo2YqUS/view?usp=sharing

BryanYu1909, 2020-08-05:
Hello, do you get a solution, I am facing a similiar issue

MichaelGrupp, 2020-10-23:
You should tune local SLAM first, it drifts a lot. And maybe then adapt the loop closure search radius.

closed 2020-10-23 by MichaelGrupp