cartographer-project/cartographer_ros #1346: error running Deutsches Museum demo

Gurtaj, 2019-07-01:
I am following the Cartographer ROS integration page to install Cartographer. The build and install ran without any errors but when I try to run the demo bag it gives the following error 👍 

Command : roslaunch cartographer_ros demo_backpack_2d.launch bag_filename:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag

Error : [demo_backpack_2d.launch] is neither a launch file in package [cartographer_ros] nor is [cartographer_ros] a launch file name
The traceback for the exception was written to the log file.

Any suggestions ?

mgladkova, 2019-07-02:
Have you sourced setup.bash file in your catkin workspace?

e.g. source ~/catkin_ws/devel/setup.bash

gsh89, 2019-07-02:
Yes I did. It still gives the same error.

carl1218, 2019-07-02:
If you had sourced setup.bash,you can try run
 `source ~/catkin_ws/devel/setup.bash`
than run
`rospack profile`

closed 2019-07-03 by gsh89