cartographer-project/cartographer_ros #1349: Problem compiling cpp including Cartographer (cannot find lua.hpp)

UserUsingGit, 2019-07-04:
Hello, I am having a problem compiling this file https://gist.github.com/ojura/85507cd6c8be419357305b30d9bb949d found in issue #965 to export trajectories from a .pbstream file. This is my CMakeLists.txt:
[CMakeLists.txt](https://github.com/googlecartographer/cartographer_ros/files/3359339/CMakeLists.txt)

Cartographer does not seem to find the lua.hpp, even if I have the library installed.
The errormessage I get is the following:

Scanning dependencies of target export_trajectories
[ 50%] Building CXX object CMakeFiles/export_trajectories.dir/export_trajectories.cpp.o
In file included from /home/usr/catkin_ws/install_isolated/include/cartographer/common/lua_parameter_dictionary.h:25:0,
                 from /home/usr/catkin_ws/install_isolated/include/cartographer/mapping/pose_graph.h:26,
                 from /home/usr/repos/cart_ros_programs/export_trajectories.cpp:3:
/home/usr/catkin_ws/install_isolated/include/cartographer/common/lua.h:20:10: fatal error: lua.hpp: No such file or directory
 #include <lua.hpp>
          ^~~~~~~~~
compilation terminated.
CMakeFiles/export_trajectories.dir/build.make:62: recipe for target 'CMakeFiles/export_trajectories.dir/export_trajectories.cpp.o' failed
make[2]: *** [CMakeFiles/export_trajectories.dir/export_trajectories.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/export_trajectories.dir/all' failed
make[1]: *** [CMakeFiles/export_trajectories.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

The command "sudo apt-get install liblua5.2-dev" gives the following output

Reading package lists... Done
Building dependency tree       
Reading state information... Done
liblua5.2-dev is already the newest version (5.2.4-1.1build1).
The following packages were automatically installed and are no longer required:
  linux-headers-4.18.0-22 linux-headers-4.18.0-22-generic linux-image-4.18.0-22-generic linux-modules-4.18.0-22-generic linux-modules-extra-4.18.0-22-generic
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 220 not upgraded.

So the dependency should be properly installed.
Does anyone know whats the problem and how to solve it?

UserUsingGit, 2019-07-05:
I figured out the mistake and came up with a dirty solution. 
The first problem was that I did not use the catkin build tool by ROS. The second problem (that I evaded but did not solve) was that several dependencies to cartographer + cartographer_ros where missing in my CMakeLists.txt. In the end I just integrated the above code into the cartographer_ros package as an additional function in pbstream_map_publisher_main.cc and build the workspace using 
catkin_make_isolated --install --use-ninja
If I find some time (and if anyone needs it) I will seperate the function into another file (e.g. trajectories_to_txt.cc) and upload it.

closed 2019-07-05 by UserUsingGit

artemiialessandrini, 2019-10-09:
Hi, @UserUsingGit , I would be happy to have some advisory, maybe get your edited CMakeLists file.
Cannot figure out adapting as a lame cmaker:)

UserUsingGit, 2019-10-10:
Hi dude, its been a while, but as far as I remember I did all the modifications in 
 the "cartographer\src\cartographer_ros\cartographer_ros\cartographer_ros\" directory.

I edited the CMakeLists.txt and created a new cpp file, both shown here: https://gist.github.com/UserUsingGit/807a2ed347d8497930918248d1ce0439

Then I rebuilt Cartographer and used the node with a command like:
cartographer_pbstream_trajectory_writer -pbstream_filename /path/to/pbstream
(I am not sure if this is the exact command, you might have to play around with it)
Gotta run cartographer first to create the pbstream.

Let me know if it works for you!