cartographer-project/cartographer_ros #1350: Lying trajectory messages

alankdkd, 2019-07-08:
I'm subscribing to the /trajectory_node_list topic for the 2D Cartographer Backpack demo, but the markers I'm getting back are lying about the position/orientation.  (Related issue: #1340) 

I know the messages are real because they don't start until I run the 2D demo, and they stop when I stop the demo, but I get the same incorrect position/orientation **each time**:

> [ INFO] [1562609378.742887695, 1432648929.447202104]: Got a message!
> [ INFO] [1562609378.743888043, 1432648929.447202104]: Size = 4
> [ INFO] [1562609378.744081143, 1432648929.447202104]: Position[i]: 0: 0, 0, 0.05
> [ INFO] [1562609378.744265161, 1432648929.447202104]: Orientation[i]: 0: 0, 0, 0, 1
> [ INFO] [1562609378.746352749, 1432648929.447202104]: Position[i]: 1: 0, 0, 0.05
> [ INFO] [1562609378.746522630, 1432648929.447202104]: Orientation[i]: 1: 0, 0, 0, 1
> [ INFO] [1562609378.746954695, 1432648929.447202104]: Position[i]: 2: 0, 0, 0.05
> [ INFO] [1562609378.747106731, 1432648929.447202104]: Orientation[i]: 2: 0, 0, 0, 1
> [ INFO] [1562609378.747600954, 1432648929.447202104]: Position[i]: 3: 0, 0, 0.05
> [ INFO] [1562609378.747753689, 1432648929.447202104]: Orientation[i]: 3: 0, 0, 0, 1

The actual position/orientation in the 2D demo are constantly changing.  Why is cartographer_ros lying to me??

Here's the code:

```
#include <iostream>
#include <ros/ros.h>
#include <visualization_msgs/Marker.h>
#include <visualization_msgs/MarkerArray.h>
#include <geometry_msgs/Pose.h>

using namespace visualization_msgs;
using namespace geometry_msgs;
using namespace std;

ros::Subscriber sub;

void poseMessageReceived (const visualization_msgs::MarkerArray& arr)  {
       ROS_INFO_STREAM( "Got a message!\n" );

         Marker m = arr.markers[0];     // First marker.
         int size = arr.markers.size();
         ROS_INFO_STREAM("Size = " << size << "\n");

         for (int i = 0;   i < size;   ++i)
         {
                 m = arr.markers[i];

                 Pose p = m.pose;
                 Point pos = p.position;
                 Quaternion ori = p.orientation;

                 ROS_INFO_STREAM("Position[i]: " << i << ": " << pos.x << ", " << pos.y << ", " << pos.z << "\n");
                 ROS_INFO_STREAM("Orientation[i]: " << i << ": " << ori.x << ", " << ori.y << ", " << ori.z << ", " << ori.w << "\n");
         }
 }

int main (int argc, char **argv)  {
        // Initialize the ROS system and become a node.
        ros::init (argc, argv, "subscribe_to_pose");
        ros::NodeHandle nh;
        ROS_INFO_STREAM("We're in main.\n");

      // Create a subscriber object.
      //ORIG: ros::Subscriber sub = nh.subscribe("cartographer_node/trajectory_node_list", 1000,
      sub = nh.subscribe("/trajectory_node_list", 1000,
                            poseMessageReceived);
      if (!sub)
        ROS_INFO_STREAM("nh.subscribe() returned nullptr!\n");

            // Let ROS take over.
            ros::spin();

        ROS_INFO_STREAM("Back from ros::spin().\n");
}
```

Package.xml:
```
<?xml  version ="1.0"?>
<package>
        <name>agitr</name>
        <version>0.0.1</version>
        <description>
                Examples from A Gentle Introduction to ROS
        </description>
<maintainer  email="alan@h3dgamma.com">
        Alan
</maintainer>
<license>TO DO</license>
<buildtool_depend>catkin</buildtool_depend>

<!--
  <build_depend>std_msgs</build_depend>
  <run_depend>std_msgs</run_depend>
-->

<build_depend>geometry_msgs</build_depend>
<run_depend>geometry_msgs</run_depend>

<build_depend>roscpp</build_depend>
<run_depend>roscpp</run_depend>

<run_depend>visualization_msgs</run_depend>
<run_depend>cartographer_ros</run_depend>

find_package(catkin REQUIRED COMPONENTS roscpp cartographer_ros visualization_msgs)
</package>
```

CMakeLists.txt:

```
# What version of CMake is needed?
cmake_minimum_required(VERSION 2.8.3)

# Name of this package.
project (agitr)

# Find the catkin build system, and any other packages on
# which we depend.
find_package(catkin REQUIRED COMPONENTS roscpp geometry_msgs visualization_msgs cartographer_ros)

# Declare our catkin package.
catkin_package ()

# Specify locations of header files.
include_directories(include ${catkin_INCLUDE_DIRS})

# Declare the executable , along with i t s source f i l e s.  I f
# there are multiple executables, use multiple copies of
# this line.
add_executable (subpose subpose.cpp)

# Specify l i b r a r i e s against which to l i n k.  Again , t h i s
# l i n e should be copied f o r each d i s t i n c t executable in
# the package.
target_link_libraries(subpose ${catkin_LIBRARIES})
```

alankdkd, 2019-07-10:
After some study, it's probably a coordinate-frame problem.  If the default coordinate frame is on the robot/sensor, it would produce the constant poses that are being produced.  Unfortunately there's no obvious way to transform the poses I'm receiving to the map frame.  Of course it's _possible_, I'm just seeing a huge disconnect between the tutorials and the task I'm trying to accomplish.

alankdkd, 2019-07-10:
The key to doing the transformation is getting the transformation from the listener:

`listener.lookupTransform("/map", "/base_link", ros::Time(0), transform);`

Then you subtract the transform's origin from the position:

```
 double x = pos.x - transform.getOrigin().x();
 double y = pos.y - transform.getOrigin().y();
```

Of course there are "gotchas" that get in the way:

1. Package.xml needs:

<build_depend>tf</build_depend>
<run_depend>tf</run_depend>

2. CMakeLists.txt needs:

find_package(catkin REQUIRED COMPONENTS roscpp geometry_msgs visualization_msgs cartographer_ros tf)

with the tf package added to the list.

3. When you try to look up the transform, it fails mysteriously.  You have to wait until it's ready:

        listener.waitForTransform("/map", "/base_link", ros::Time(0), ros::Duration(3.0));
        listener.lookupTransform("/map", "/base_link", ros::Time(0), transform);

This gives you the position of the robot along the trajectory.  Next: Getting the orientation.  I'll post the complete solution after I figure it out.

alankdkd, 2019-07-10:
Finally solved it.  This code gives the poses along the trajectory as the Backpack 2D demo runs/publishes.

The source:

```
#include <iostream>
#include <cmath>
#include <ros/ros.h>
#include <visualization_msgs/Marker.h>
#include <visualization_msgs/MarkerArray.h>
#include <geometry_msgs/Pose.h>
#include <tf/transform_listener.h>

using namespace visualization_msgs;
using namespace geometry_msgs;
using namespace std;

ros::Subscriber sub;

void poseMessageReceived (const visualization_msgs::MarkerArray& arr)  {
         ROS_INFO_STREAM( "Got a message!\n" );

        tf::TransformListener listener;
        tf::StampedTransform transform;
        int size = arr.markers.size();
        ROS_INFO_STREAM("Size = " << size << "\n");

        listener.waitForTransform("/map", "/base_link", ros::Time(0), ros::Duration(3.0));
        listener.lookupTransform("/map", "/base_link", ros::Time(0), transform);

         for (int i = 0;   i < size;   ++i)
         {
                Marker m = arr.markers[i];

                Pose p = m.pose;
                Point pos = p.position;
                Quaternion ori = p.orientation;

                ROS_INFO_STREAM("Position[i]: " << i << ": " << pos.x << ", " << pos.y << ", " << pos.z << "\n");
                ROS_INFO_STREAM("Orientation[i]: " << i << ": " << ori.x << ", " << ori.y << ", " << ori.z << ", " << ori.w << "\n");
                ROS_INFO_STREAM("Transform x y: " << transform.getOrigin().x() << ", " << transform.getOrigin().y() << "\n");

                double x = pos.x - transform.getOrigin().x();
                double y = pos.y - transform.getOrigin().y();
                double angle = atan2(y, x);
                ROS_INFO_STREAM("x = " << x << ", y = " << y << ", angle = " << angle << "\n");
         }
 }

int main (int argc, char **argv)  {
        // Initialize the ROS system and become a node.
        ros::init (argc, argv, "subscribe_to_pose");

        ros::NodeHandle nh;
        ROS_INFO_STREAM("We're in main.\n");
        ros::Rate rate(10.0);

        // Create a subscriber object.
        sub = nh.subscribe("/trajectory_node_list", 1000, poseMessageReceived);
        if (!sub)
                ROS_INFO_STREAM("nh.subscribe() returned nullptr!\n");

        // Let ROS take over.
        ros::spin();
        ROS_INFO_STREAM("Back from ros::spin().\n");
}
```
Package.xml:

`<?xml  version ="1.0"?>
<package>
        <name>agitr</name>
        <version>0.0.1</version>
        <description>
                Examples from A Gentle Introduction to ROS
        </description>
<maintainer  email="alan@h3dgamma.com">
        Alan
</maintainer>
<license>TO DO</license>
<buildtool_depend>catkin</buildtool_depend>

<build_depend>tf</build_depend>
<run_depend>tf</run_depend>

<build_depend>geometry_msgs</build_depend>
<run_depend>geometry_msgs</run_depend>

<build_depend>roscpp</build_depend>
<run_depend>roscpp</run_depend>

<run_depend>visualization_msgs</run_depend>
<run_depend>cartographer_ros</run_depend>
</package>
`

CMakeLists.txt:

` What version of CMake is needed?
cmake_minimum_required(VERSION 2.8.3)

# Name of this package.
project (agitr)

# Find the catkin build system, and any other packages on
# which we depend.
find_package(catkin REQUIRED COMPONENTS roscpp geometry_msgs visualization_msgs cartographer_ros tf)

# Declare our catkin package.
catkin_package ()

# Specify locations of header files.
include_directories(include ${catkin_INCLUDE_DIRS})

# Declare the executable , along with i t s source f i l e s.  I f
# there are multiple executables, use multiple copies of
# this line.
add_executable (subpose subpose.cpp)

# Specify l i b r a r i e s against which to l i n k.  Again , t h i s
# l i n e should be copied f o r each d i s t i n c t executable in
# the package.
target_link_libraries(subpose ${catkin_LIBRARIES})
`

alankdkd, 2019-07-11:
For completeness, here's the 3D version.  The main difference is orientation angle can be computed from x, y, and the arc tangent in 2D, but must be converted from the quaternion in 3D.  Also the 3D backpack demo runs much slower than 2D, so the duration for the waitForTransform() call was increased to 30 seconds.  (And it still times out eventually; need to look into this.)

`[subpose.txt](https://github.com/googlecartographer/cartographer_ros/files/3383584/subpose.txt)`

alankdkd, 2019-07-11:
Just noticed the 2D angle doesn't match the yaw angle, suggesting there's a bug somewhere.  FYI.

alankdkd, 2019-07-15:
There are still some loose ends to this issue...

1. My code gives the poses in real time, but after loop closure, the poses are optimized to minimize error by global SLAM.   I want to obtain these more-accurate poses.
2. Ojura mentioned (in #1340) that the TrajectoryQuery service in Cartographer will provide trajectory poses.  This may be how to accomplish issue 1.
3. As mentioned above, there's an inconsistency between the 2D angle in the xy plane, and the yaw angle obtained from the quaternion.

If anyone has any answers to these issues, feel free to comment.  I'll post answers as I figure them out.

Joar93, 2019-09-26:
Hi @alankdkd! Tried your code and it worked for me! Thank you!
I will workout a way to export the values to a txt. document for further analysis.

MichaelGrupp, 2020-10-14:
You should use the `/trajectory_query` service or get the live poses from TF. You are trying to use the marker topic which does not have the orientations and should only be used for visualization purposes.

closed 2020-10-14 by MichaelGrupp